espressif/arduino-esp32 #884: abort() was called at PC 0x40103e5a on core 1

Dreadout, 2017-11-27:
### Hardware:
Board: Wemos LOLIN32					
Core Installation/update date:
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 115200

### Description:
I am using RFID to scan cards and I want to send the UID of the card to a database trough PHP.
I can scan the card and connect to the internet but only when the program comes to the database connection part it crashes.

Any ideas ?

### Sketch:
```cpp

#include <SPI.h>
#include <MFRC522.h>
#include <WiFi.h>

#define RST_PIN         22          // Configurable, see typical pin layout above
#define SS_PIN          5         // Configurable, see typical pin layout above
int bstate = 0;
String txData = "";
int rfid = 0;
int status = WL_IDLE_STATUS; 

WiFiClient client;

char server[] = "192.168.0.240";
char ssid[] = "myinternet";
char pass[] = "mysuppersecretpass";

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance
String  UID;
void setup() {
	Serial.begin(115200);		// Initialize serial communications with the PC
	SPI.begin();			      // Init SPI bus
	mfrc522.PCD_Init();		 // Init MFRC522
	mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
  
Serial.println("Attempting to connect to WPA network...");
status = WiFi.begin(ssid, pass);

if(status != WL_CONNECTED) {
  delay(500);
  Serial.println("Connected to network");
}

else{
  Serial.println("Coudln't get a wifi connection");
}
  
	Serial.println(F("Scan PICC to see UID, SAK, type, and data blocks..."));
}

void loop() {
	// Look for new cards
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	 
 Serial.print("Printing HEX UID : ");

 String rfid = "";
    for (byte i = 0; i < mfrc522.uid.size; i++) {
        rfid += mfrc522.uid.uidByte[i] < 0x10 ? "0" : "";
        rfid += String(mfrc522.uid.uidByte[i], HEX);
    }
  Serial.print(rfid);
  Serial.println("");
  txData = "rfid="+ (String (rfid));
  delay(1000);
  exit(0);
}
void DataToMySQL(){

  if (client.connect(server, 3306)) {
    Serial.println("--> connection ok\n");
    client.print("GET /write_data.php?"); // This
    client.print("value="); // This
    client.print(txData); // And this is what we did in the testing section above. We are making a GET request just like we would from our browser but now with live data from the sensor
    client.println(" HTTP/1.1"); // Part of the GET request
    client.print( "Host: " );
    client.println(server);
    client.println("Connection: close"); // Part of the GET request telling the server that we are over transmitting the message
    client.println(); // Empty line
    client.println(); // Empty line
    client.stop();    // Closing connection to server
    Serial.println("--> finished transmission\n");   
  }

  else{
  Serial.println("Connection to database failed");
  Serial.println();

}
}

```

### Serial Monitor
My serial monitor is returning this.
```
Attempting to connect to WPA network...
Connected to network
Scan PICC to see UID, SAK, type, and data blocks...
Printing HEX UID : 9e5dc249 (card is scanned here)
abort() was called at PC 0x40103e5a on core 1

Backtrace: 0x40087560:0x3ffd6380 0x4008765f:0x3ffd63a0 0x40103e5a:0x3ffd63c0 0x400d5033:0x3ffd63e0 0x400d13ab:0x3ffd6400 0x4011a82c:0x3ffd6450

Rebooting...

```

copercini, 2017-11-27:
>  exit(0);

Why are you using this?

Dreadout, 2017-11-27:
``
exit(0);
``
Is deleted. Program works now tho I can't seem to get the UID of the scan into my database.

me-no-dev, 2017-11-28:
you have a web server running on port 3306? And constructing a request like this is a really bad idea ;)
You are in effect sending a really small chunks of data every time you call client.print. Also ```value=rfid=9e5dc249``` <<< not a good value ;) needs escaping.
You can do something like:
```cpp
client.printf("GET /write_data.php?value=%s HTTP/1.0\r\nHost: %s\r\nConnection: close\r\n\r\n", txData.c_str(), server);
```