espressif/arduino-esp32 #885: ESP32 Thing Freezes after invoking Wifi function

osamaqalam, 2017-11-27:
### Hardware:
Board:							ESP32 Thing Dev Module
Core Installation/update date:			27/Nov/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
The micro controller stops execution after the first statement that uses a function from the wifi library.

### Sketch:
`void connectToWiFi(const char * ssid, const char * pwd)
{
  int ledState = 0;

  printLine();
  Serial.println("Connecting to WiFi network: " + String(ssid));

  WiFi.persistent(false) ; 
  WiFi.begin(ssid, pwd);

  unsigned long timeout = millis();
  while (WiFi.status() != WL_CONNECTED) 
  {
    // Blink LED while we're connecting:
    // digitalWrite(LED_PIN, ledState);
    ledState = (ledState + 1) % 2; // Flip ledState
    delay(500);
    Serial.print(".");

  if ( millis() - timeout > wifiTimeout)
      return;
  }

  Serial.println();
  Serial.println("WiFi connected!");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}`

### Debug Messages:
```
------------------------------
Connecting to WiFi network: sm-a520w7319
[D][WiFiGe "Garbage letters"
```

stickbreaker, 2017-11-27:
Are you sure you can call `persistent()` before the `begin()`?

Chuck

osamaqalam, 2017-11-27:
@stickbreaker 
Yes, I used the following
`  WiFi.persistent(false) ; `

stickbreaker, 2017-11-27:
So, when you call `WiFi.persistent(false)` the ESP32 hangs?  If that is the case, I would call it after it the `begin()`

Chuck.

osamaqalam, 2017-11-27:
I tried that and same behavior occurred.
Is there a way to debug the wifi module to check if it is functioning correctly or not?

stickbreaker, 2017-11-27:
Can you try the example code (WiFiClient.ino)?  Does that work?

Chuck.

osamaqalam, 2017-11-27:
The code worked twice at the beginning then gave same behavior of freezing afterwards.

stickbreaker, 2017-11-27:
A lot of the weird errors are related to power,  when the WiFi section is working it can burst above 400mA.  Brown outs cause bad stuff to happen, How are you powering your board?

Chuck.

osamaqalam, 2017-11-27:
From the USB port on the laptop.

everslick, 2017-12-02:
I also suspect the PSU. Even using a different connector or cable can make subtle differences.

osamaqalam, 2017-12-02:
Using another ESP32 thing without soldering it to a protoboard fixed the issue.
Since this behavior only occurred for MCUs that were soldered onto a protoboard, it might be an issue that happened due to bad soldering. 
Thanks for the help.

closed 2017-12-02 by osamaqalam