espressif/arduino-esp32 #886: EPS32 RTC Clock restarts when EN goes low and high again

helmut64, 2017-11-27:
I am trying to understand how an RTC clock can be used as a continues RTC clock source even if the EN line goes to low and high again, which is the implementation of the ESP32 development board reset switch.

I reading the TIME0-LOW and TIME1-HIGH registers and they reset and start from scratch instead of continuing the RTC. Using deep sleep, the RTC continues as expected, however using a push on the EN button it resets.

Is this the normal behaviour?
Any workarounds?

### Hardware:
Board:   ESP32 Dev Module and Heltec ESP32 with RTC crystal.

me-no-dev, 2017-11-28:
totally normal. When you hit the EN pin you power off the chip and all is reset.

closed 2017-11-28 by me-no-dev

helmut64, 2017-11-28:
Thank you for your feedback, this is what I was seeing. With the older ESP8266 there was a reset pin to overcome this, however for users of the ESP32 they should not disable the enable, and never click on the EN button.

me-no-dev, 2017-11-28:
You can set deep sleep to wake up from a pin and go to sleep as soon as you are done. That way the ESP will be sleeping until you wake it up and then go back to sleep. Isn't that what you are looking for?

helmut64, 2017-11-28:
You are right, I do this wakeup PIN already, however every time somebody clicks on "EN" the RTC time is gone. The same when uploading a new file which causes EN to go to zero.

I believe the only way to program the ESP32 without destroying the RTC is to do a wakeup and PULL IO0 to low, and do an ESP.reset().

me-no-dev, 2017-11-28:
if EN was not clearing RTC, how would you clear it otherwise? To program the ESP without losing RTC, maybe you need to OTA?