espressif/arduino-esp32 #887: Bootloader executable name mismatch

lonerzzz, 2017-11-27:
### Hardware:
Board:							NA
Core Installation/update date:		2017/11/27
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Just noticed a problem with the Arduino IDE use of the bootloader where the binary for flashing the image does not match the expected name. The application expected boot_loader_qio.bin but the actual versions were bootloader_qio_40m.bin and bootloader_qio_80m.bin. When I copied bootloader_qio_80m.bin to bootloader_qio.bin, everything worked.

This seems related to the issue just solved under https://github.com/espressif/arduino-esp32/pull/881.

me-no-dev, 2017-11-28:
Arduino IDE fine on my side. OS? There was a miss with Linux but I fixed that too. Did not reboot IDE after update?

lonerzzz, 2017-12-07:
Didn't get a notification for this. Just saw it now. I am on Windows 10. The error happens reboot or not. Is there supposed to be file link or something?

me-no-dev, 2017-12-11:
seems your platform.txt and the sdk files are desynced somehow. https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/bin

lonerzzz, 2017-12-21:
@me-no-dev Seems a persistent problem in my case as I just pulled the code down once again.

everslick, 2018-04-19:
Should we still keep this open?

closed 2019-08-15 by stale[bot]