espressif/arduino-esp32 #888: Adafruit ESP32 Feather hangs at init when calling SPIFFS.begin()

Kestranor, 2017-11-28:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			2017.11.24
IDE name:							Arduino IDE
Flash Frequency:					80MHz
Upload Speed:						921600

### Description:
Calling SPIFFS.begin() at startup causes the ESP32 to hang there. The program is stuck at that step forever. At first, I had an SPIFFS mount failed error, then tried calling SPIFFS.begin(true) instead as per a suggestion here. After this, I get no "mount failed" errors, but the program execution hangs indefinitely. Disabling this call lets the program run again (with no file system active, obviously).

### Sketch:
void FS_Initialize(void)
{
	/* Mount SPIFFS. */
	if (!SPIFFS.begin(true))
	{
#ifdef DEBUG_MODE
		Serial.println("Failed to mount SPIFFS.");
#endif
		spiffs_mounted = false;
	}
	else
	{
#ifdef DEBUG_MODE
		/* Mount successful: list all files in debug mode. */
		Serial.println("SPIFFS mount succeeded.");
		FS_ListFiles("/", 0);
#endif
		spiffs_mounted = true;

#ifdef FS_TEST_MODE
		TestFS();
#endif
	}
}

NOTE: I get no serial prints either here. It just hangs.
NOTE 2: FS_Initialize is called in setup()

MarkyAD, 2017-11-28:
How long did you let it run? Calling SPIFFS.begin(true) formats the filesystem, which can take a while to run. After formatting once, call SPIFFS.begin() without any parameters to just mount the filesystem. Better yet, load a test sketch to do the formatting, then load the sketch you're working on.

Kestranor, 2017-11-28:
I let it run for around 5-10 minutes. I assumed that would be enough. Do I need to let it run longer?

MarkyAD, 2017-11-28:
Dont have a feather to test but I just ran a format on a lolin32 and it was faster than I thought.. like under a minute. Maybe try a simple test sketch with just serial debugging and your FS code, which looks ok to me (besides the one typo...)

/ Mount successful: list all files in debug mode. */

needs // or /*

beegee-tokyo, 2017-11-28:
I am using PlatformIO and I had a weird problem with SPIFFS. The partition for SPIFFS was not initialized and SPIFFS.begin(true) or SPIFFS.format() didn't work at all. 
However that was months ago and I do not remember how I solved it. But maybe you can google for it.

me-no-dev, 2017-11-28:
I ran similar sketch yesterday on my feather and had no issues to format or mount spiffs. Which board have you selected?

Kestranor, 2017-11-28:
I have selected Adafruit ESP32 Feather.
I also tried googling for it and while my google-fu is usually stronk, I found no real answer.

beegee-tokyo, 2017-11-28:
I remember now how I solved it.
In ArduinoIDE I went to menu TOOLS->ESP32 Sketch Data Upload.
I asked me 
No files have been found in your data folder!
Are you sure you want to create an empty SPIFFS image?
I selected yes and it uploaded an empty SPIFFS image to my ESP32. After that I had no more problems with SPIFFS.

Kestranor, 2017-11-28:
@beegee-tokyo: Thanks for the suggestion. Just tried it, uploading the empty image "succeeds", but unfortunately, it still hangs at begin in the code.

me-no-dev, 2017-11-28:
@Kestranor whoa??? how is that possible? maybe bad flash? Can you switch to ESP32 Dev Board and set debug level to Verbose and give it another go?

MarkyAD, 2017-11-28:
I thought for sure fixing the two unbalanced comments would have solved it, unless he already fixed those. From the original post, it starts with /* Mount SPIFFS. / and ends with / Mount successful: list all files in debug mode. */ making that whole section in between one big comment (which is why the compiler didnt barf on it), that section even shows up in italics (maybe github is playing with my head again, but no beer tonight).

me-no-dev, 2017-11-28:
@MarkyAD there :) I fixed the sketch code to display properly ;)

MarkyAD, 2017-11-28:
oh ok yes I can see now said the blind man. :)

MarkyAD, 2017-11-28:
I dunno I tested the same code and I couldnt get it NOT to work.

Kestranor, 2017-11-28:
@me-no-dev Changed it, does not help. Should I attach the build log?

me-no-dev, 2017-11-28:
wat does the serial log say? Upload seems fine

Kestranor, 2017-11-28:
Funnily enough I uploaded a file via the ESP32 Flash tool again, switched back to the Adafruit ESP32 Feather again, re-flashed the SPIFFS test sketch and now it seems to work fine. Have to try flashing my own sketch again to see if it also does, but at least it shows signs of life where it did not before. Thanks all for the input and assistance. I'll mark this as closed.

closed 2017-11-28 by Kestranor