espressif/arduino-esp32 #892: Now able to update the internal time of ESP-32

parmanjosan, 2017-11-29:
Hi,
I want to update the internal time of ESP-32 with Arduino ide basically I want that when ESP-32 have wifi connection then it updates its internal time and when it doesn't have the net connection then it updates it's internal store time and prints the time after every 10min.

My board is: nodemcu-32s
Time Zone - India(UTC+05:30)
plz help me.

me-no-dev, 2017-11-29:
```cpp
#include <WiFi.h>
#include "time.h"

const char* ssid       = "user";
const char* password   = "pass";
const char* ntp_server = "pool.ntp.org";
const int   hourOffset = 1;

void printLocalTime()
{
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
        Serial.println("Failed to obtain time");
        return;
    }
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup()
{
    Serial.begin(115200);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    
    configTime(hourOffset * 3600, 3600, ntp_server);
    printLocalTime();

    WiFi.disconnect(true);
    WiFi.mode(WIFI_OFF);
}

void loop()
{
  delay(1000);
  printLocalTime();
}
```

closed 2017-11-29 by me-no-dev

parmanjosan, 2017-11-30:
Thanks, Dav but I want to update time in deep sleep mode.

stickbreaker, 2017-11-30:
@parmanjosan in deep sleep mode, the processor is not running? How do you expect the processor to update time if it is not running?

Do you want to keep an accurate time value while the processor sleeps?  If so, you will need a RTC that stays awake an counts time.  like a DS1307 Or you will have to provide a 32khz crystal and write ULP code.

Chuck.

parmanjosan, 2017-11-30:
thanks, Mr. stickbreaker. Yes that's what I want to do, I need Arduino ide code which updates RTC clock value.

schweini, 2018-04-21:
I'd just like to clarify for posterity that the above mentioned code DOES keep accurate time during deep sleep, as far as I can see.

configTime() seems to forget the offsets, though. But timestamps should be in UTC, anyhow, IMHO.