espressif/arduino-esp32 #893: esp32-hal-log.h macros not usable by default

atanisoft, 2017-11-29:
### Hardware:
Board:						ESP32 Dev Module
Core Installation/update date:	29/Nov/2017
IDE name:					Platform.io
Flash Frequency:				40Mhz
Upload Speed:				115200

### Description:
The macros defined in esp32-hal-log.h are not usable due to sdkconfig.h having CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL set to 0 (NONE). It would be good if this could be configured at project level.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <esp32-hal-log.h>

void setup() {
  Serial.begin(115200L);
  log_e("test output - error");
  log_w("test output - warning");
  log_i("test output - info");
  log_d("test output - debug");
  log_v("test output - verbose");
}

void loop() {
}
```

In the above the "test output" lines will not be displayed on serial. The only solution I have found is modify sdkconfig.h to comment out the definition of CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL and pass this in via -DCONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=XXXX in platformio.ini, very likely the Arduino IDE will behave similarly.

stickbreaker, 2017-11-30:
these debug levels can be set at compile by selecting "ESP32 Dev Module" as your  board under tools,
then another option will appear "Core Debug Level" . 

If you want to add these options to your specific board, in boards.txt
```
*.menu.DebugLevel.none=None
*.menu.DebugLevel.none.build.code_debug=0
*.menu.DebugLevel.error=Error
*.menu.DebugLevel.error.build.code_debug=1
*.menu.DebugLevel.warn=Warn
*.menu.DebugLevel.warn.build.code_debug=2
*.menu.DebugLevel.info=Info
*.menu.DebugLevel.info.build.code_debug=3
*.menu.DebugLevel.debug=Debug
*.menu.DebugLevel.debug.build.code_debug=4
*.menu.DebugLevel.verbose=Verbose
*.menu.DebugLevel.verbose.build.code_debug=5
```
Replace the `*` with your board's id.  I am using 
>"Wemos" Wifi&Bluetooth Battery

```
##############################################################

WeMosBat.name="WeMos" WiFi&Bluetooth Battery

WeMosBat.upload.tool=esptool
WeMosBat.upload.maximum_size=1310720
WeMosBat.upload.maximum_data_size=294912
WeMosBat.upload.wait_for_upload_port=true

WeMosBat.serial.disableDTR=true
WeMosBat.serial.disableRTS=true

WeMosBat.build.mcu=esp32
WeMosBat.build.core=esp32
WeMosBat.build.variant=pocket_32
WeMosBat.build.board=Pocket32

WeMosBat.build.f_cpu=240000000L
WeMosBat.build.flash_mode=dio
WeMosBat.build.flash_size=4MB
WeMosBat.build.boot=bootloader
WeMosBat.build.partitions=default

WeMosBat.menu.FlashFreq.80=80MHz
WeMosBat.menu.FlashFreq.80.build.flash_freq=80m
WeMosBat.menu.FlashFreq.40=40MHz
WeMosBat.menu.FlashFreq.40.build.flash_freq=40m

WeMosBat.menu.UploadSpeed.921600=921600
WeMosBat.menu.UploadSpeed.921600.upload.speed=921600
WeMosBat.menu.UploadSpeed.115200=115200
WeMosBat.menu.UploadSpeed.115200.upload.speed=115200
WeMosBat.menu.UploadSpeed.256000.windows=256000
WeMosBat.menu.UploadSpeed.256000.upload.speed=256000
WeMosBat.menu.UploadSpeed.230400.windows.upload.speed=256000
WeMosBat.menu.UploadSpeed.230400=230400
WeMosBat.menu.UploadSpeed.230400.upload.speed=230400
WeMosBat.menu.UploadSpeed.460800.linux=460800
WeMosBat.menu.UploadSpeed.460800.macosx=460800
WeMosBat.menu.UploadSpeed.460800.upload.speed=460800
WeMosBat.menu.UploadSpeed.512000.windows=512000
WeMosBat.menu.UploadSpeed.512000.upload.speed=512000

WeMosBat.menu.DebugLevel.none=None
WeMosBat.menu.DebugLevel.none.build.code_debug=0
WeMosBat.menu.DebugLevel.error=Error
WeMosBat.menu.DebugLevel.error.build.code_debug=1
WeMosBat.menu.DebugLevel.warn=Warn
WeMosBat.menu.DebugLevel.warn.build.code_debug=2
WeMosBat.menu.DebugLevel.info=Info
WeMosBat.menu.DebugLevel.info.build.code_debug=3
WeMosBat.menu.DebugLevel.debug=Debug
WeMosBat.menu.DebugLevel.debug.build.code_debug=4
WeMosBat.menu.DebugLevel.verbose=Verbose
WeMosBat.menu.DebugLevel.verbose.build.code_debug=5

##############################################################
```

Chuck.

atanisoft, 2017-11-30:
@stickbreaker unfortunate PlatformIO does not have a tools menu and I have "esp32dev" as the board type. Even if I was using the Arduino IDE, how does the sdkconfig.h get updated? Does it override it during the build somehow?

stickbreaker, 2017-11-30:
Yea, when those options are selected, they are passed into the build as command line args. 

I don't know PlatformIO maybe @me-no-dev, can direct you.

Chuck.

atanisoft, 2017-11-30:
@stickbreaker can you post an example of the commandline? I can try and convert it to PlatformIO.

stickbreaker, 2017-11-30:
Here is from my latest compile.  I think you want `-DCORE_DEBUG_LEVEL=0`

```
Compiling library "tinyFS"
"C:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_Pocket32 -DARDUINO_ARCH_ESP32  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\variants\pocket_32" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\SPI\src" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src" "-IC:\Users\user\Documents\Arduino\libraries\RTClib" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\FS\src" "-IC:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\SD\src" "-IC:\Users\user\Documents\Arduino\libraries\OneWire" "-IC:\Users\user\Documents\Arduino\libraries\tinyFS\src" "C:\Users\user\Documents\Arduino\libraries\tinyFS\src\tinyFS.cpp" -o "C:\Users\user\AppData\Local\Temp\arduino_build_25231\libraries\tinyFS\tinyFS.cpp.o"
```

in platform.txt:
```
# This can be overriden in boards.txt
build.flash_size=4MB
build.flash_mode=dio
build.boot=bootloader
build.code_debug=0
build.extra_flags=-DESP32 -DCORE_DEBUG_LEVEL={build.code_debug}
```

is this what you want?

Chuck.

beegee-tokyo, 2017-11-30:
@atanisoft 
In PlatformIO put into platformio.ini file:
**`build_flags = -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG`** or whatever debug level you want.

atanisoft, 2017-11-30:
Thanks @stickbreaker and @beegee-tokyo. That does seem to work, is there a way to split it up so the core logging is at a different level than say application code that also uses log_X?

beegee-tokyo, 2017-11-30:
@atanisoft 
I don't think its possible to split the log level between core and app but I am not the expert.

atanisoft, 2017-11-30:
@beegee-tokyo I think you are right.  The ESP-IDF logging API has a "tag" which can be used to set specific log levels for each tag, unfortunately the HAL log wrappers are discarding that tag in the ESP_LOGE and other macros. I am guessing without a bit of work on the HAL layer to expose the tag/log level/etc pieces from the lower levels it won't work cleanly.

One other aspect I haven't been able to sort out with PlatformIO is enabling CONFIG_ARDUHAL_LOG_COLORS.  The serial monitor seems to just print the ANSI codes rather than interpret them. I am guessing this is likely a setting in PlatformIO that is incorrect as well as a missing configuration in the Windows Power Shell used by PlatformIO. Any thoughts on that?

beegee-tokyo, 2017-11-30:
@atanisoft regarding CONFIG_ARDUHAL_LOG_COLORS I think it is depending on the serial terminal program you use. I never use the PlatformIO terminal but instead I use Termite. But that doesn't support coloring as well. It just suppresses the color codes.

atanisoft, 2017-11-30:
@beegee-tokyo PlatformIO uses MiniTerm IIRC. I fired up MTPuTTY and the colors work there :) The only downside of an external terminal is having to remember to close it before uploading but that is minor.

beegee-tokyo, 2017-11-30:
@atanisoft will try MiniTerm IIRC. Maybe you can close this issue now.
@me-no-dev maybe this is worth to label as "tip" or "help" or something. I would really like to see a Wiki for this repo where solved issues like this are collected in a FAQ. I am willing to help to set it up and maintain.

atanisoft, 2017-11-30:
@beegee-tokyo I agree, this should likely get closed as it is not an issue but certainly this should be documented somewhere.

closed 2017-11-30 by atanisoft

az-z, 2020-07-20:
hate to resurface the issue, but based on the code line 139 https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-log.h , the setting of CONFIG_ARDUHAL_ESP_LOG=1 should direct the execution to idf log.h.
but how do i  set CONFIG_ARDUHAL_ESP_LOG ? // i do not have arduino ide installed.

NarinLab, 2021-03-30:
Any update on Platform IO ?

ankostis, 2022-04-07:
@atanisoft to view color logs, you must instruct MiniTerm to display chars received in *raw*, with this in your `platformio.ini`:
```ini
[env]
monitor_filters = direct    ; to view colored logs, if enabled with `-DCONFIG_ARDUHAL_LOG_COLORS` or `
```

Alternatively, launch the monitor command directly from your OS's terminal (if working with a POSIX host):
```bash
pio device monitor`
```