espressif/arduino-esp32 #898: Side effect in Serial 0 with HSPI port

jestebanes, 2017-12-03:
Hardware:

Board:	Own design based on ESP-WROOM-32
Core Installation/update date:	30/Nov/2017
IDE name:	Arduino IDE
SDK version: v3.0-dev-1325-ge5b2c1cf
Flash Frequency:	80Mhz
Upload Speed:	512000

Description:
I've got my own design based in ESP-WROOM and I'm using HSPI port to connect an external 64M flash memory with a library that works fine with ESP8266.
I've got the issue with Serial port 2 and I can't change the baud rate on serial port 2. Then I realize that I can change the pinout and use Serial port 0 with external pins 16 & 17 and put the Serial 2 for debug purpose on pins 3 & 1.
But I found a new extrange behabiour on ESP32 and when I change the buad rate on Serial port 0, using the commands
Serial.end();
Serial.begin(newbaudrate, SERIAL_8N1, 16, 17);
The HSPI port stops to work ¿? and I lost the connection with the external flash. I try to use the VSPI port instead but with the same result.

Kind regards

me-no-dev, 2017-12-11:
I discussed this with some of our engineers and none seem to be able to come up with a reason to why would that happen. HSPI and UART are separate peripherals. Will need to look into this further

Vorms, 2018-01-05:
I am very concerned by this bug and I need to change the speed of the UART2.
Thanks to keep me in touch and inform me when a workaround is made.
Best regards.

closed 2019-08-15 by stale[bot]