espressif/arduino-esp32 #899: WiFi.status() WL_CONNECT_FAILED  no fired when password in invalid

mytrain, 2017-12-03:
**### Hardware:**
Board:"Wemos" WiFi Battery
Core Installation/update date: 29/Nov/2017
IDE name: Arduino IDE
SDK Version: v3.0-dev-1325-ge5b2c1cf

**### Description:**
Trying to connect to a SSID with an invalid password does not switch WiFi.Status() to WL_CONNECT_FAILED (WiFi.Status() remains unchanged and equals to WL_DISCONNECTED) event after reception of WiFi event SYSTEM_EVENT_STA_DISCONNECTED.

me-no-dev, 2017-12-04:
can you try the esp32 dev module as board selected and also raise the debug level in the board menu? Should show what events are actually received

lonerzzz, 2017-12-14:
@mytrain Have you had a chance to try the suggestion?

mytrain, 2017-12-17:
Hi
Sorry for the delay but I was moving in a new home, ... no much time to play with my ESP

@me-no-dev  I switch to ESP32 Dev Module with debug level set to Verbose but nothing changed

So I made a short sketch to reproduce the issue (in attachment but with TXT extension as INO is not accepted).
[issue899.txt](https://github.com/espressif/arduino-esp32/files/1565958/issue899.txt)
 Here is the result :

[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
>>>WIFI SYSTEM_EVENT_STA_START
6
6
[....]
6
6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 2 - AUTH_EXPIRE
>>>WIFI SYSTEM_EVENT_STA_DISCONNECTED
6
6
....

I would expected WiFi.status() switches to WL_CONNECT_FAILED  on AUTH_EXPIRE event.

Regards

closed 2019-08-15 by stale[bot]