espressif/arduino-esp32 #900: Reboot loop with latest version

mitchsf, 2017-12-04:
Windows 10, ESP32 DevKitC

With the latest core just installed on a new pc, working programs reboot in a loop. The first line hopefully indicates the issue. That is not a path or user on my pc. 

Thanks in advance for help with this.

assertion "res == pdTRUE" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c", line 368, function: start_cpu0_default
abort() was called at PC 0x400dde57 on core 0

Backtrace: 0x40088168:0x3ffe3b90 0x40088267:0x3ffe3bb0 0x400dde57:0x3ffe3bd0 0x40082c7a:0x3ffe3c00 0x40082e24:0x3ffe3c30 0x4007891e:0x3ffe3c50 0x40078987:0x3ffe3c80 0x40078aca:0x3ffe3cb0 0x40078bf2:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0

mitchsf, 2017-12-04:
Exception decoder:

Decoding 11 results
0x40088168: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x40088267: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 572
0x400dde57: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40082c7a: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 368 (discriminator 1)
0x40082e24: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 210

me-no-dev, 2017-12-04:
can you get it down to minimal reproducible sketch? What features are you using? Exception seem to happen really early

mitchsf, 2017-12-04:
I'm working on it. Sketch is failing somewhere in Setup()

me-no-dev, 2017-12-04:
from the backtrace above, it does not even get to setup :) probably out of heap

me-no-dev, 2017-12-04:
stack sorry

mitchsf, 2017-12-04:
Removed all code in loop and setup, and it doesn't fail. With setup code, it goes back to a reset loop. I'll work to figure out where it's failing.

mitchsf, 2017-12-04:
Removing sections of code clears the issue, but any one of several sections, so possibly that does indicate a problem with the stack. Going back to the old pc with a version not more than 2-4 weeks old, the original sketch compiles and runs perfectly. 

Is there a way to further debug this from here? Can the stack size be increased?

me-no-dev, 2017-12-04:
depends which one :) can you please post some code that shows the issue? I need something to work with

stickbreaker, 2017-12-04:
@me-no-dev are Global variables allocated in the task's Stack space?  Maybe @mitchsf has a few large structures pre-allocated that fill Stack.  The Default Arduino task has a 8k stack?

Chuck.

mitchsf, 2017-12-04:
I'm trying to put something together that crashes, without having to deal with some modified libraries.

mitchsf, 2017-12-04:
Here is a mess of code that compiles, and hopefully demonstrates the problem. 

One hint - when ledcWrite() is removed from ledIndicator(), the sketch runs successfully.

`//#include <SPI.h>
//#include <WiFi.h>
//#include <NTPClientOne.h>
//#include <TimeZoneOne.h>
//#include <TimeLib.h>
#include <TinyGPS++.h>
//#include <WiFiUdp.h>
//#include <Preferences.h>
//#include <Bounce.h> // use the original Bounce library included with Bounce2
//#include <Adafruit_Sensor.h>
//#include <Adafruit_BME280.h>
//#include <SimpleTimer.h>
//#include <FastLED.h>

//#define DEBUG

hw_timer_t * timer = NULL;
hw_timer_t * toneTimer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

const byte fRev = 1; // firmware revision

//WiFiServer server(80);
//WiFiClient client;
//WiFiUDP ntpUDP;
//IPAddress Ip(192, 168, 13, 1); // ip address for AP

const int udpSyncInterval = 60000; // NTP sync every 10 minutes (ms)
const int timeSyncInterval = 30; // seconds to sync with time server (seconds)

// You can specify the time server pool and the offset, (in seconds)
// additionaly you can specify the update interval (in milliseconds).
// NTPClientOne timeClient(ntpUDP, "europe.pool.ntp.org", 3600, 60000);
//NTPClientOne timeClient(ntpUDP, "us.pool.ntp.org", 0, udpSyncInterval);

HardwareSerial Serial1(1);
#define gpsRx 13
#define gpsTx 2

#define serialGPS Serial1
TinyGPSPlus gps;

struct tmt { // used for digit change effects
  byte hour10;
  byte hour1;
  byte minute10;
  byte minute1;
  byte second10;
  byte second1;
  byte tenth;
  byte hundredth;
};
tmt currTime;
tmt oldTime;
tmt *switchTime;

// initialize and instantiate tz object
//TimeZoneOne tz;
//TimeChangeRule dSt; // = {};
//TimeChangeRule sTd; // = {};

// set variables for FastLED
#define DATA_PIN 15
#define CLOCK_PIN 4
#define COLOR_ORDER BGR
#define LED_TYPE APA102
[... 502 lines cut ...]
void clockReset(byte pattern) {

}
/*----------------------------------------------------------------------*
   Weather sensor BME280
  ----------------------------------------------------------------------*/
unsigned long int weather(byte sFunction, byte scale) {
 
  return 0;
}

/*----------------------------------------------------------------------*
   Blink LED
  ----------------------------------------------------------------------*/
void blinkLED(byte blinkRate) {

}

/*----------------------------------------------------------------------*
   stepNixies() used at startup
  ----------------------------------------------------------------------*/
void stepNixies() {

}
`

mitchsf, 2017-12-09:
Any updates on this? I went back to a two week old version, and everything works. Thanks.

rhyslbw, 2018-03-06:
Have you tried reflashing using esptool `erase_flash`?

everslick, 2018-04-19:
@mitchsf Is this issue still relevant or did you find a solution in the meantime? Also please use the correct MARKDOWN syntax for code snippets (three ticks before and after source code).

daduke, 2018-05-29:
commit 600f4c4130f2411e8ff33396d0ef40a7ec257cd6 broke it.

mitchsf, 2018-05-29:
Sorry, I have not tried it again with a newer release.

mitchsf, 2019-01-02:
I just tried with the latest release, and the problem persists. Sketch works perfectly using an older version. 

The sketch uses 25% of program storage space, and 13% of dynamic memory.

Below is the latest result from the exception decoder.

configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
assertion "err == ESP_OK" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/crosscore_int.c", line 93, function: esp_crosscore_int_init
abort() was called at PC 0x400df637 on core 0

Backtrace: 0x40090464:0x3ffe3b80 0x40090667:0x3ffe3ba0 0x400df637:0x3ffe3bc0 0x400e068d:0x3ffe3bf0 0x400836fc:0x3ffe3c10 0x400838e0:0x3ffe3c40 0x4007892f:0x3ffe3c60 0x40078995:0x3ffe3c90 0x400789a0:0x3ffe3cc0 0x40078b33:0x3ffe3ce0 0x40080335:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Decoding stack results
0x40090464: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x40090667: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x400df637: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400e068d: esp_crosscore_int_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/crosscore_int.c line 93
0x400836fc: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 350
0x400838e0: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/cpu_start.c line 213

me-no-dev, 2019-01-02:
sounds like memory issue probably to fail that early @mitchsf . Please post some code

lbernstone, 2019-01-03:
Memory management is not a simple subject, but, if you have a 32k struct, it will need contiguous space.  The earlier you set that up in your code, the less likely your SRAM is to be fragmented.  If the memory will not be frequently accessed, you could get a WRover module, and store it in PSRAM.

mitchsf, 2019-01-04:
Thanks for the help!

Marc-Aurele, 2019-01-29:
Hi,

I had the same issue. It stopped rebooting when i launched make erase_flash command.

closed 2019-08-15 by stale[bot]