espressif/arduino-esp32 #902: Serial port not showing up on Linux

paramono, 2017-12-06:
### Hardware:
Board:							WeMos WiFi&Bluetooth Battery
IDE name:						Arduino IDE v1.8.5
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
My issue is most likely a problem with my setup, not with the arduino-esp32 itself, but I still need some help.

I am trying to program my ESP32 board from a Linux system. I followed all the instructions - cloned the repo, recursively updated submodules and ran `python get.py`, everything went successfully. 

When I launch Arduino IDE v1.8.5, I can select the board from `Tools -> Board` option, but `Tools -> Port` option is grayed out. I guess I am missing something in my setup.

To clarify: my user is added to `dialout` group, and if use some other board (for example, Arduino Nano), the port option becomes configurable.

What should I do make port configurable for ESP32? Which serial device should appear in my system?

### Sketch:

```cpp
// The sketch is pretty much irrelevant to the problem, but being able to upload
// and run the classic "blink" program will help clarify that everything works

int ledPin = 16;

void setup() {
    pinMode(ledPin, OUTPUT);
}

void loop() {
    digitalWrite(ledPin, HIGH);
    delay(500);
    digitalWrite(ledPin, LOW);
    delay(500);
}
```

copercini, 2017-12-06:
My guess is that Silabs cp2102 drivers are missing: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

paramono, 2017-12-06:
@copercini 

I downloaded, compiled and inserted the cp2102 module into my Linux kernel. Unfortunately, it didn't change much, as ttyUSB0 still doesn't appear in my system (even after logout).

From your link:

>If device driver does not install automatically, identify USB to serial converter chip on your ESP32 board (or external converter dongle), search for drivers in internet and install them.

Frankly, the chip that looks like the USB to serial converter chip has nothing written on it, so I cannot identify which exact driver I need. I tried downloading FTDI Virtual COM port drivers - namely, copied shared objects to my system, but didn't help.

I have ESP-WROOM-32, the board is "WeMos WiFi&Bluetooth Battery" if that helps.
Any suggestions on where I should look further?

paramono, 2017-12-06:
I am sorry for reporting this issue. Turns out, my USB-microUSB was bad. I replaced it, and then my ESP32 showed up in the system and I managed to successfully upload my sketch.

Closing the issue.

closed 2017-12-06 by paramono

marcocrio, 2019-09-05:
I'm so glad this happened to you!  had the exact same problem, I tried everything, reinstalled the drivers and everything. Turns out my cable was bad as well. It was driving me crazy

adilbaig, 2021-04-04:
I have the same problem with an ESP32-WROVER board as well. (Board name: 'TTGO T8 V1.1 ESP32 4MB PSRAM TF CARD 3D ANTENNA WiFi&bluetooth ESP32-WROVER Micropython' from [AliExpress](https://www.aliexpress.com/item/4001066527709.html))

Looking at the comments above I tried 2 different cables. Both times, a blue LED lights up and a green LED, near the TXD pin, starts blinking. But the board does not show up as `/dev/ttyUSB0` (or `/dev/ttyUSB1`) under my machine (Ubuntu 20.04). In fact, plugging it in does not show ANY new device.

I have verified that I have the cp210 drivers via `lsmod`

```
> lsmod | grep cp210
cp210x                 32768  0
usbserial              53248  1 cp210x
```
The user is also a member of the `dialout` group. 

```
> groups
.. tty uucp dialout cdrom sudo dip plugdev lpadmin sambashare docker libvirt
```

I tried resetting the board by pressing the RESET button. Nothing works. Absolutely no idea what's wrong.

adilbaig, 2021-04-04:
Same issue with the cable. While my cables were good (4th was the charm!), there's a difference between MicroUSB cables capable of transmitting data vs those that only charge. There's no way to tell by looking at it. I used my PS4 game controller's charging cable to get this to work.

theisuru, 2021-07-31:
I would have spent so much time to figure out this problem, if not have came accross this thread. Thanks guys. I also had the same problem with the USB cable.

Spockwurst, 2022-08-29:
I had the same problem an I was super confused. Thank you!

capaci, 2022-12-13:
Had the same issue here. Thank you @paramono

maayanhaimdulberg, 2023-10-26:
> I am sorry for reporting this issue. Turns out, my USB-microUSB wire was bad. I replaced it, and then my ESP32 showed up in the system and I managed to successfully upload my sketch.
> 
> Closing the issue.

thanks that solved my problem hehe

Cpreet, 2024-04-10:
Same problem, thanks for the solution

Xitee1, 2024-09-15:
I've had the same problem but it was not caused by a bad cable in my case.
It was `brltty` interfering with it (found it out by using `dmesg` after connecting it). In my case disabling or masking `brltty` + rebooting didn't help, it was still somehow taking control over my ESP. However after purging it, my ESP finally showed up in the list!