espressif/arduino-esp32 #903: delay doesn't work for small values like delay(1)

jpmonettas, 2017-12-06:
### Description:
Describe your problem here

### Sketch:
```cpp
unsigned long start_millis=millis();
unsigned long timeout=5000;
while(timeout-- > 0){
    delay(1);
}
Serial.println(millis()-start_millis);

// prints 60 instead of 5000
```
This breaks libraries that use this technique like 
https://github.com/adafruit/Adafruit_FONA/blob/master/Adafruit_FONA.cpp#L1756-L1790

stickbreaker, 2017-12-06:
@jpmonettas 
The `delay(1)` is treated as a special case, the FreeRTOS task switcher treats it a `yield()` call, It passes this task's time slice to then next task with equal priority.  If there are no task with equal priority it returns to this task without executing the delay.  That is what you are seeing. You might try `delay(2)` as a work-a-round. I don't know if espressif will consider handling this case. 

Chuck.

Dannsei, 2018-01-29:
`delayMicroseconds(1000)  `works

sub-millisecond values also work. e.g. `delayMicrosceconds(380)`   

But the issue is that some libraries are getting broken because `delay(1) `is not working., and the libarries are using `delay(1)`.  I don't know how` delay(1)` can be redefined as `delayMicroseconds(1000)` in the ESP32 core. Perhaps the easy fix is to submit a pull request to the lrelevant ibraries with this substituion to make the library ESP32 friendly.

akotowski, 2019-05-24:
This is still a problem, also for value of 2, not just 1

closed 2019-08-14 by stale[bot]