espressif/arduino-esp32 #904: License - no license to use

smitthhyy, 2017-12-06:
More a product bug than a code bug (bug being an unintended outcome of a design).

I am keen to use this code but have run stuck with copyright and licensing. As there is no license included, the only thing I am allowed to do is to view it and fork it. I can't use or modify the code in any way. See https://opensource.guide/legal/. I don't believe this to be the authors intent as evident by their terms and conditions on their website (http://espressif.com/en/content/terms-use section 8).

Can I suggest including a LICENSE file in the root with the following MIT license text (without the tags):

<MITLicense>
Copyright <YEAR> <COPYRIGHT HOLDER>

Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"), to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
</MITLicense>

This MIT license allows the greatest usage flexibility which is why it's become the most common open source license today.  Note, please do not use a GNU license as these are quite restrictive for many hence their popularity has greatly declined.

Thankyou for your assistance.

### Hardware:
Board:							N/A
Core Installation/update date:			7Dec2017
IDE name:							N/A
Flash Frequency:					N/A
Upload Speed:						N/A

### Description:
See above text

### Sketch:
N/A

### Debug Messages:
N/A

RudyFiero, 2017-12-07:
What exactly is the problem with the following notice?

>  This library is free software; you can redistribute it and/or
>  modify it under the terms of the GNU Lesser General Public
>  License as published by the Free Software Foundation; either
>  version 2.1 of the License, or (at your option) any later version.
>  This library is distributed in the hope that it will be useful,
>  but WITHOUT ANY WARRANTY; without even the implied warranty of
>  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
>  Lesser General Public License for more details.
>  You should have received a copy of the GNU Lesser General Public
>  License along with this library; if not, write to the Free Software
>  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

smitthhyy, 2017-12-07:
Not sure what your question refers to, so guessing either:
1.	Where is this displayed?  Ie it’s not in the root of the repository so doesnt apply to this repository. 
2.	Or if you are asking what is wrong with the GNU Lesser GPL, the code of the application is something that cannot be provided in certain cases which is a breach of 4d0 or 4d1. Unfortunatally that also means not using the Espressif hardware too. 

As said, I don’t believe this to be Espressif intent to put restrictions in place therefore encourage them to use the **Espressif MIT license** as used in their ESP8266 RTOS SDK @ https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/LICENSE

smitthhyy, 2017-12-08:
That's not completely correct. The one of interest that lead to this issue is ArduinoOTA. No license exists from the root down to the library itself. Many of the examples also don't have licenses so are not usable. Were the repository doesn't have a license, any discussions are not free, ie they are copyright with very limited restrictions on use.

devyte, 2017-12-08:
For the record, this repo doesn't seem to have a global license like the esp8266/Arduino repo does. There, right above the green button that says Clone or download, there is an LGPL-2.1 link, while in this repo it's missing.

tynx, 2018-05-03:
Any updates regarding a repo-wide/general license?

ryanmitts, 2019-03-05:
I'm also interested in what the license is of this. I noticed one of the files in LGPL in this library.

closed 2019-08-15 by stale[bot]

atanisoft, 2019-08-15:
@me-no-dev can you add a LICENSE file to the repo to satisfy this still outstanding issue (even if it went stale and got closed due to inactivity by filer)