espressif/arduino-esp32 #906: how to use Serial event in esp32 arduino ide

jt040-jetnet, 2017-12-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:

Hello all
i want to receive some data serially on my esp32 
for that i want to enable uart receive interrupt ,
in arduino serialEvent() is used for receive interrupt ,but its not working for esp32
below is code    available in examples of arduino ide 
can any one suggest me how to receive data in interrupt for esp32 Arduino IDE

### Sketch:

String inputString = "";         // a String to hold incoming data
boolean stringComplete = false;  // whether the string is complete

void setup() {
  // initialize serial:
  Serial.begin(9600);
  // reserve 200 bytes for the inputString:
  inputString.reserve(200);
}

void loop() {
  // print the string when a newline arrives:
  if (stringComplete) {
    Serial.println(inputString);
    // clear the string:
    inputString = "";
    stringComplete = false;
  }
}

/*
  SerialEvent occurs whenever a new data comes in the hardware serial RX. This
  routine is run between each time loop() runs, so using delay inside loop can
  delay response. Multiple bytes of data may be available.
*/
void serialEvent() {
  while (Serial.available()) {
    char inChar = (char)Serial.read();
    inputString += inChar;
    if (inChar == '\n') {
      stringComplete = true;
    }
  }
}

shaielc, 2018-01-07:
hi haven't seen an implementation of the serial event anywhere ,so i kinda added it for my project no clue if it is stable (just made it), but it works for me at the moment. This is a "bandage" solution I didn't go deep into the idf to find out how it should have been done.

https://gist.github.com/shaielc/28a94b39dbd232974a6d78e384d44b0f

rappleto, 2018-02-02:
Thanks very much, @shaielc.
To apply this the hard way, I took the four source files from the zip, and manually applied the changes that you had made to the same files in my ~arduino-1.8.5/hardware/espressif/esp32/cores/esp32 folder. 
I ended up using the highly intuitive vimdiff for the compares.
I learned about IRAM_ATTR for interrupt routines - important safety tip, thank-you.
In this case, the interrupt routine assignment from your serialEventExample.ino is set explicitly. If I understand correctly, this is a true interrupt routine as opposed to the Arduino target implicit calls to serialEvent() which just happen after every loop() cycle.

shaielc, 2018-02-03:
This just adds your interrupt function after the handling of the original uart interrupt from the idf.
glad to help!

Vorms, 2018-08-03:
Hello, many thanks for this article.
I would like to know if it is possible to declare the input buffer "inputString" in a class or not.
My goal is to "share" the input string with a class. I have to examine the string in a class.
Any example will be welcome !
Best regards
Thierry

rappleto, 2019-01-10:
From what I gather, SPI uses a different driver from this UART serialevent one. There is some documentation here, though I didn't see much that explains how to register and use the SPI interrupt routine:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html
and further, the use of interrupt driven SPI seems a bit bizarre in that instead of handling the interrupt, the docs suggest you poll spi_device_get_trans_result() to see if the interrupt driven transactions are complete.
The API reference shows int spicommon_irqsource_for_host(spi_host_device_t host) which might allow you to tap into the IRQ instead of polling the result from the driver.

rappleto, 2019-01-10:
As a starting point you could try running this:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_master/main/spi_master_example_main.c

rappleto, 2019-01-10:
> Hello, many thanks for this article.
> I would like to know if it is possible to declare the input buffer "inputString" in a class or not.
> My goal is to "share" the input string with a class. I have to examine the string in a class.
> Any example will be welcome !
> Best regards
> Thierry

You can declare it in a class, but as a static member. But I don't see why you would need to examine it in a class, or if you do want to you can certainly access an unclassed variable from within a class. This was tried out on an ATmega328p using the Arduino IDE.

It shows a global inputString accessed by an object/instance method: parseInput(), and by a static class method: printStrings() as well as by a non-class method: setup().
For kicks I also added this. The static class String classyStaticString is also accessed by those same three methods.

String inputString = "";

class cInputParser
{
  public:
  void parseInput();
  static String classyStaticString;
};

// Note: without this the classyStaticString will not be found by the linker
String cInputParser::classyStaticString;

class cStringPrinter
{
  public:
  static void printStrings();
};

cInputParser ip;

void cInputParser::parseInput()
{
  char first = inputString.charAt(0);

  classyStaticString = inputString;

  switch (first)
  {
    case 'a':
      inputString = "b";
      break;
    case 'b':
      inputString = "c";
      break;
    default:
      inputString = "a";
      break;
  }
}

void cStringPrinter::printStrings()
{
  Serial.print("classyStaticString=");
  Serial.print(cInputParser::classyStaticString);
  Serial.print(" inputString=");
  Serial.println(inputString);
}

void setup()
{
  // put your setup code here, to run once:
  Serial.begin(115200);
  cInputParser::classyStaticString = "something";
  inputString = "b";
  cStringPrinter::printStrings();
}

void loop()
{
  // put your main code here, to run repeatedly:
  ip.parseInput();
  cStringPrinter::printStrings();
  delay(1000);
}

closed 2019-08-15 by stale[bot]

yottayuan, 2019-10-18:
> hi haven't seen an implementation of the serial event anywhere ,so i kinda added it for my project no clue if it is stable (just made it), but it works for me at the moment. This is a "bandage" [...]

could u share the code?

yottayuan, 2019-10-18:
> hi haven't seen an implementation of the serial event anywhere ,so i kinda added it for my project no clue if it is stable (just made it), but it works for me at the moment. This is a "bandage" [...]

hi, could u share the code? the link cannot be getted to .

shaielc, 2019-10-20:
it's set as public but i attached it anyway:

[28a94b39dbd232974a6d78e384d44b0f-f913c78955815b7ddf1f496740c25a44ccd6934a.zip](https://github.com/espressif/arduino-esp32/files/3747827/28a94b39dbd232974a6d78e384d44b0f-f913c78955815b7ddf1f496740c25a44ccd6934a.zip)

arehmansoomro, 2020-07-15:
I'm getting this error when I test the serialEventExample on ESP32, what am I missing:

serialEventExample:9:11: error: 'class HardwareSerial' has no member named 'setInterrupt'
   Serial1.setInterrupt(&SerialEvent1);
           ^
serialEventExample:9:25: error: 'SerialEvent1' was not declared in this scope
   Serial1.setInterrupt(&SerialEvent1);
                         ^
exit status 1
'class HardwareSerial' has no member named 'setInterrupt'

mrfibreoptic, 2021-01-27:
Could you please edit this .ino please, thanks:

#include <Wire.h>                //enable I2C.
#define address 99               //default I2C ID number for EZO pH Circuit.

char computerdata[20];           //we make a 20 byte character array to hold incoming data from a pc/mac/other.
byte received_from_computer = 0; //we need to know how many characters have been received.
byte serial_event = 0;           //a flag to signal when data has been received from the pc/mac/other.
byte code = 0;                   //used to hold the I2C response code.
char ph_data[20];                //we make a 20 byte character array to hold incoming data from the pH circuit.
byte in_char = 0;                //used as a 1 byte buffer to store inbound bytes from the pH Circuit.
byte i = 0;                      //counter used for ph_data array.
int time_ = 815;                 //used to change the delay needed depending on the command sent to the EZO Class pH Circuit.
float ph_float;                  //float var used to hold the float value of the pH.

void setup()                     //hardware initialization.
{
  Serial.begin(9600);           //enable serial port.
  Wire.begin();                 //enable I2C port.
}

void serialEvent() {                                                              //this interrupt will trigger when the data coming from the serial monitor(pc/mac/other) is received.
  received_from_computer = Serial.readBytesUntil(13, computerdata, 20);           //we read the data sent from the serial monitor(pc/mac/other) until we see a <CR>. We also count how many characters have been received.
  computerdata[received_from_computer] = 0;                                       //stop the buffer from transmitting leftovers or garbage.
  serial_event = true;                                                            //set the serial event flag.
}

void loop() {                                                                     //the main loop.
  if (serial_event == true) {                                                     //if a command was sent to the EZO device.
    for (i = 0; i <= received_from_computer; i++) {                               //set all char to lower case, this is just so this exact sample code can recognize the "sleep" command.
      computerdata[i] = tolower(computerdata[i]);                                 //"Sleep" ≠ "sleep"
    }
    i=0;                                                                          //reset i, we will need it later 
    if (computerdata[0] == 'c' || computerdata[0] == 'r')time_ = 815;             //if a command has been sent to calibrate or take a reading we wait 815ms so that the circuit has time to take the reading.
    else time_ = 250;                                                             //if any other command has been sent we wait only 250ms.
    

    Wire.beginTransmission(address);                                              //call the circuit by its ID number.
    Wire.write(computerdata);                                                     //transmit the command that was sent through the serial port.
    Wire.endTransmission();                                                       //end the I2C data transmission.

    if (strcmp(computerdata, "sleep") != 0) {  	                                  //if the command that has been sent is NOT the sleep command, wait the correct amount of time and request data.
                                                                                  //if it is the sleep command, we do nothing. Issuing a sleep command and then requesting data will wake the circuit.

      delay(time_);								                                                //wait the correct amount of time for the circuit to complete its instruction.

      Wire.requestFrom(address, 20, 1); 		                                      //call the circuit and request 20 bytes (this may be more than we need)
      code = Wire.read();               		                                      //the first byte is the response code, we read this separately.

      switch (code) {							          //switch case based on what the response code is.
        case 1:                         		//decimal 1.
          Serial.println("Success");    		//means the command was successful.
          break;                        		//exits the switch case.

        case 2:                         		//decimal 2.
          Serial.println("Failed");     		//means the command has failed.
          break;                        		//exits the switch case.

        case 254:                       		//decimal 254.
          Serial.println("Pending");    		//means the command has not yet been finished calculating.
          break;                        		//exits the switch case.

        case 255:                       		//decimal 255.
          Serial.println("No Data");    		//means there is no further data to send.
          break;                        		//exits the switch case.
      }

      while (Wire.available()) {         		//are there bytes to receive.
        in_char = Wire.read();           		//receive a byte.
        ph_data[i] = in_char;					      //load this byte into our array.
        i += 1;                          		//incur the counter for the array element.
        if (in_char == 0) {              		//if we see that we have been sent a null command.
          i = 0;                         		//reset the counter i to 0.
          break;                         		//exit the while loop.
        }
      }

      Serial.println(ph_data);          		//print the data.
    }
    serial_event = false;                   //reset the serial event flag.
  }
  //Uncomment this section if you want to take the pH value and convert it into floating point number.
  //ph_float=atof(ph_data);
}

mrfibreoptic, 2021-01-27:
Could you please edit this for me :)
[pH_I2C.ino.zip](https://github.com/espressif/arduino-esp32/files/5877653/pH_I2C.ino.zip)

ghilanakram, 2021-06-13:
> ### Hardware:
> Board: ESP32 Dev Module
> Core Installation/update date: 11/jul/2017
> IDE name: Arduino IDE
> Flash Frequency: 40Mhz
> Upload Speed: 115200
>
> ### Description:
> Hello all
> i want to receive some [...]
use **_serialEventRun()_** instead of **_serialEvent()_**

`void serialEventRun(){    
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag so the main loop can
    // do something about it:
    if (inChar == '\n') {
      stringComplete = true;
    }
  }
}`

rappleto, 2021-06-13:
There is an example here - have to scroll down to near the bottom for a 
working solution:

https://esp32.com/viewtopic.php?t=9939

Attaching signature
🍁🍎 </home/rappleto/.thunderbird/AppleFlagSignature.png>