espressif/arduino-esp32 #907: ESP32 Rebooting and abort() infinite loop

HQ5, 2017-12-08:
Hi I was using Lolin 32 Lite for my project and I discovered that if I use wifi connection in Lolin32, some of the pins won't work.

I couldn't find a full clear list of pins which are available or not in the same discussion so I tried to test it out by printing every value of the pin in the serial.

I just defined every pin as input and print the value in the serial while I connect 3V output of the Lolin32 to each pin.

While testing the values stopped being updated and when I reset the board with the physical rst button, I got an infinite loop of the following text in the serial :

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0010,len:4
load:0x3fff0014,len:588
load:0x40078000,len:0
load:0x40078000,len:9768
entry 0x400789b8
abort() was called at PC 0x400d282e on core 0

Backtrace: 0x40087048:0x3ffca110 0x40087147:0x3ffca130 0x400d282e:0x3ffca150 0x400f8fe0:0x3ffca170 0x400f9ee9:0x3ffca190 0x400f9fc3:0x3ffca200 0x400fa2e5:0x3ffca260 0x400f8b95:0x3ffca2c0 0x400f8853:0x3ffca310 0x400f889f:0x3ffca330 0x400d1698:0x3ffca350 0x4010d96f:0x3ffca370 0x400d9b77:0x3ffca3a0

I tried to understand the error by using the ESP32 Exception Debugger and I got the followings:

0x40087048: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 546
0x40087147: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 546
0x400d282e: operator new(unsigned int) at /home/ivan/e/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc line 54
0x400f8fe0: nvs::HashList::insert(nvs::Item const&, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_item_hash_list.cpp line 57
0x400f9ee9: nvs::Page::mLoadEntryTable() at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 579
0x400f9fc3: nvs::Page::load(unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_page.cpp line 76
0x400fa2e5: nvs::PageManager::load(unsigned int, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_pagemanager.cpp line 27
0x400f8b95: nvs::Storage::init(unsigned int, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_storage.cpp line 241
0x400f8853: nvs_flash_init_custom at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 350
0x400f889f: nvs_flash_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/nvs_flash/src/nvs_api.cpp line 350
0x400d1698: initArduino at C:\Users\DELL\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-misc.c line 83
0x4010d96f: app_main at C:\Users\DELL\Documents\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 24
0x400d9b77: main_task at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 349

I didn't have any in-depth education of programming so I can't understand what is the problem...

Thank you for helping me out in advance.

### Hardware:  
Board:	Lolin32 Lite						?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

me-no-dev, 2017-12-11:
which pins exactly did you set to input? all GPIOs?

HQ5, 2017-12-11:
I don't understand why but I somehow fixed it by long pressing the reset button on the board !

closed 2017-12-11 by HQ5

bhavish030, 2019-05-27:
@HQ5  Hello. I am having the same error, and i can;t fix this. I tried long pressing on the reset button(EN) but it doesn't work. Do you know if there is any other alternative to this please? Thank you