espressif/arduino-esp32 #908: light_sleep in v3.0-dev has several issues

tferrin, 2017-12-09:
### Hardware:
Board:							Adafruit Huzzah32
Core Installation/update date:			6/dec/2017 (v3.0-dev-1325-ge5b2c1cf)
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80MHz
Upload Speed:						115200

### Description:
There are several problems with the current implementation of light_sleep.  Since these are all related I'm including them together in one ticket.

Synopsis:
1) A timer-based light_sleep sometimes never wakes up.  The WDT never triggers, but the call to esp_light_sleep_start() never returns.
2) The millis() timer doesn't tick during light_sleep leading to values that have nothing to do with the current run time of the sketch.  It is easy to correct the value returned by millis() in user code, but a better place to do this is in the underlying implementation of millis().
3) When light_sleep is made to work for longer time periods (see details below), the getimeofday() and time() clocks tick too fast.

Details:
1)  timer-based light_sleep sometimes never wakes up: calling suspend() in the sketch below will eventually cause esp_light_sleep_start() to never return.  Small values of sleep (e.g., 10ms) will usually cause the hang to happen within a few minutes.  Larger values (e.g., 1000ms) may take a few hours before the hang occurs.  To me, this behavior is indicative of a race condition in esp_light_sleep_start() that is "tickled" the more frequently the function is called.  I have tried several fixes for this issue without success and its root cause remains elusive.

2) millis() timer doesn't tick during light_sleep: When light_sleep stalls the cpus, normal timers don't tick, including the timer used for the millis() function, and this can often lead to unexpected behavior in sketches.  The IDF provides the needed hooks to correct millis()'s notion of time (see adjustedMillis() in the sketch below), but IMHO it would be much better if the underlying implementation of millis() incorporated this timing correction.

3) light_sleep causes getimeofday() and time() to tick too fast: The output of the included sketch provides the values of several time counters each second.  Over an interval of a couple hours, these differ from real time by ~8 seconds on my WROOM-32 module.  To see this, run the sketch and compare its output with the stopwatch on your smartphone.  This behavior may be a byproduct of the different oscillator used for the RTC during light_sleep mode, but trying to re-sync the ESP32 TOD clock via NTP doesn't seem to work because the difference between the real (NTP) time and ESP32 TOD clock keeps growing.  If the TOD clock cannot be made more accurate, then at least it needs to be correctable by LWIP's sntp module.  [An example of attempting to re-sync the TOD clock via NTP is not included here but can be furnished on request.]

### Sketch:
// Note that multiple spaces don't get correctly copied & pasted and so output formatting is screwy 
#include <time.h>
#include <sys/time.h>
#include <esp_sleep.h>
#include <esp_err.h>

void suspend(uint32_t ms)
{
  uint64_t time_in_us = (uint64_t) ms * 1000;
  esp_err_t ret;

  ret = esp_sleep_enable_timer_wakeup(time_in_us);
  if (ret == ESP_OK)
    ret = esp_light_sleep_start();
  if (ret != ESP_OK)
    Serial.printf("light_sleep error: %d", ret);
}

unsigned long adjustedMillis()
{
  extern uint64_t s_microseconds_offset;    // defined in components/newlib/time.c
  static uint64_t t0_diff = 0LL;    // difference between RTC and millis() timer at T0

  if (t0_diff == 0)
    t0_diff = s_microseconds_offset;
  return millis() + ((s_microseconds_offset - t0_diff) / 1000);
}

#define TIMER_CLEAR     timerclear(&tv1); timerclear(&tv2); timerclear(&tvd)
#define TIMER_START     gettimeofday(&tv1, (struct timezone*)0)
#define TIMER_STOP      gettimeofday(&tv2, (struct timezone*)0)
#define TIMER_SUB       timersub(&tv2, &tv1, &tvd)
// time_t is an int32_t on ESP32 so need to cast before multiplying
#define TIMER_ELAPSED   ((tvd.tv_usec + ((int64_t)tvd.tv_sec * 1000000)) / 1000.0)
struct timeval tv1, tv2, tvd;
time_t now;

void setup()
{
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  // 6dec17 SDK=v3.0-dev-1325-ge5b2c1cf
  Serial.printf("\nSDK version: %s\n", ESP.getSdkVersion());
  Serial.printf(" time      ms   adjms        tod\n");
  TIMER_START;
}

void loop()
{
  static int ledON = 0;
  if (ledON)
    digitalWrite(LED_BUILTIN, LOW);
  else
    digitalWrite(LED_BUILTIN, HIGH);
  ledON ^= 1;

  time(&now);
  TIMER_STOP;
  TIMER_SUB;
  Serial.printf("%5d %7lu %7lu %10.2f\n", now, millis(), adjustedMillis(), TIMER_ELAPSED);

#if 1
  suspend(1000);
#else
  for (int i=0; i<100; i++)
    suspend(10);
#endif
}

### Debug Messages:
Sample output from the first 30 seconds of running:
SDK version: v3.0-dev-1325-ge5b2c1cf
 time      ms   adjms        tod
    0      21      21       0.06
    1      33    1030    1009.62
    2      39    2034    2013.28
    3      45    3038    3016.90
    4      50    4041    4020.51
    5      56    5044    5024.13
    6      62    6048    6027.74
    7      68    7052    7031.35
    8      73    8055    8034.97
    9      79    9058    9038.59
   10      85   10062   10042.21
   11      91   11066   11045.83
   12      96   12069   12049.47
   13     102   13073   13053.09
   14     108   14076   14056.72
   15     114   15080   15060.35
   16     120   16084   16063.98
   17     125   17087   17067.60
   18     131   18090   18071.23
   19     137   19094   19074.86
   20     143   20098   20078.49
   21     148   21101   21082.11
   22     154   22105   22085.73
   23     160   23108   23089.36
   24     166   24112   24092.99
   25     171   25115   25096.63
   26     177   26119   26100.26
   27     183   27122   27103.89
   28     189   28126   28107.52
   29     195   29130   29111.16
   30     200   30133   30114.79

helmut64, 2017-12-14:
Hi Tom, I am working also on sleep code with the ESP32, I am using as well direct programmed 64-bit timers for wakeup and time counting, here are come comments from an ESP32 beginner.

a) Check out that you are using chip revision 1 (ESP.getChipRevision())
With the 64-bit timers I had major reliability problems, they completely disappeared with rev1. Maybe the same may be true for your freezing light sleep. The good news is that rev1 fixes the long errata list, so I start my project with rev1 only.

b) 32KHz RTC crystal clock
I know that some ESP32 modules are (e.g. Heltec) using an additional 32kHz RTC clock, however I don’t know if the existence of an crystal will be detected automatically or if special SDK lib versions are needed. With the DevKit modules they use an internal RTC ossizillator and it does not work accurate.

c) Estimating sleep duration
For this I read the RTC time before sleep and on the wakeup after sleep. The elapsed time I add to my own time counting. Unfortunately this is not very accurate but better than nothing. The code is at end.

Another fact is that a reset (EN low/high) will reset the RTC to zero even a battery is connected. For more accurate time an external temperature compensated RTC chip may be needed, e.g. DS3231.

Regards Helmut

`static uint64_t ESP32_get_rtc_time_us()
{
    SET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE_M);
    while (GET_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_VALID_M) == 0) {
        ;
    }
    CLEAR_PERI_REG_MASK(RTC_CNTL_TIME_UPDATE_REG, RTC_CNTL_TIME_UPDATE_M);
    uint64_t low = READ_PERI_REG(RTC_CNTL_TIME0_REG);
    uint64_t high = READ_PERI_REG(RTC_CNTL_TIME1_REG);
    uint64_t ticks = (high << 32) | low;
    return ticks * 100 / (rtc_clk_slow_freq_get_hz() / 10000); // scale RTC_CTNL_SLOWCLK_FREQ to avoid overflow
}
`

helmut64, 2017-12-14:
Hi Tom, I was testing your sample code and got the problem you are describing. Two symptoms I have seen:

- Using your long sleep defined and changing and the 1000ms sleep into 10ms, it works fine for minutes.

- undefining your long sleep, it hangs after the first print, however adding a delay(1) after the suspend in your loop overcomes the problem.

tferrin, 2017-12-14:
I checked and the chip revision of my WROOM-32 is 1, so it's not the old version of the chip.
The Adafruit board I have does not use an external 32KHz crystal for the RTC, so an inaccurate internal oscillator could easily account for the fast ticking I've observed.
The delay(1) after after suspend is most interesting.  I'll give that a try tonight. Perhaps other RTOS tasks need a chance to run after the cpus have been stalled.

lonerzzz, 2017-12-15:
@tferrin  @helmut64 Keep in mind that the delay(1) cause a task switch rather than actually performing a delay on the ESP32 so that might be feeding another thread for you.

tferrin, 2017-12-15:
Yes. Understand.  I've added a call to yield() -- which I'm pretty sure just calls vPortYield() on the ESP32 -- immediately after esp_light_sleep_start() and my test program has been running for a few hours now, so a good sign.  Needs more testing but my fingers are crossed.

Just to be clear, my suspend() function is now...

void suspend(uint32_t ms)
{
  uint64_t time_in_us = ms * 1000LL;
  esp_err_t ret;
  ret = esp_sleep_enable_timer_wakeup(time_in_us);
  if (ret == ESP_OK) {
    ret = esp_light_sleep_start();
    yield();    // <== NEW!  
  }
  if (ret != ESP_OK) {
    log_e("light_sleep error: %d", ret);    // likely WiFi wasn't off
    delay(ms);    // if we can't stall the cpus at least we can call delay
  }
}

tferrin, 2017-12-15:
I decided to simplify the test program that calls the suspend() function posted above and not do any serial output at all.  Instead I just flash a LED every 1 second or so, then call suspend() again.  This has now been running ~6 hours without hanging!  So I'm now thinking that serial output to the USB bridge plays a role in causing the hangs I described in my first post above.

helmut64, 2017-12-16:
What I do for another MCU (STM32L4 mbed), before I go to deep sleep I wait that the serial output is flushed to avoid problems. I have not looked into the details of the serial output implementation of the ESP32 for Arduino. For the Arduino D21 (SAMD) the code it is being a synchronous output.

I also noticed that an ESP32 interrupt routing  print on the serial may hang sometimes, looks like the prints go async.

tferrin, 2017-12-17:
Okay, after doing a *lot* more testing I've reached the conclusion that the fundamental problem of the ESP32 hanging during light_sleep has nothing to do with serial I/O. I *think* it is related to calling the gettimeofday() function. I've provided a new test case below that is essentially Espressif's own component test code, except instead of calling the light_sleep function only once I call it repeatedly. As noted in my original post above, how long the program sleeps for has a direct bearing on how long this program runs before hanging. Long sleep times rarely hang, while short sleep times will hang the ESP32 within a couple minutes. The test code here will hang in an indeterminate processor state within a minute or two.

```
#include <sys/time.h>

#define TEST_ASSERT_INT32_WITHIN(delta, expected, actual) \
    if (abs(actual - expected) > delta) printf("assertation failed")

// choose one of the following...
#define VERY_SHORT_SLEEP
//#define SHORT_SLEEP
//#define LONG_SLEEP

#ifdef LONG_SLEEP
  #define SLEEP_TIME 2000000    // 2s
  #define SKIPCNT 0
#elif defined(SHORT_SLEEP)
  #define SLEEP_TIME 250000     // 250ms
  #define SKIPCNT 3
#elif defined(VERY_SHORT_SLEEP)
  #define SLEEP_TIME 10000      // 10ms
  #define SKIPCNT 75
#else
  #error "must define a sleep interval"
#endif

#define EXPECTED (SLEEP_TIME/1000)

void dotest()
{
    // code below from esp-idf-master/components/esp32/test/test_sleep.c
    // see entry TEST_CASE("wake up from light sleep using timer", "[deepsleep]")

    esp_sleep_enable_timer_wakeup(SLEEP_TIME);
    struct timeval tv_start, tv_stop;
    gettimeofday(&tv_start, NULL);
    esp_light_sleep_start();
    gettimeofday(&tv_stop, NULL);
    float dt = (tv_stop.tv_sec - tv_start.tv_sec) * 1e3f +
               (tv_stop.tv_usec - tv_start.tv_usec) * 1e-3f;
    TEST_ASSERT_INT32_WITHIN(500, EXPECTED, (int) dt);
}

void heartbeat(int skipcnt = 0)
{
  static int n = 0;
  if (skipcnt < 0) return;
  if (n++ >= skipcnt) {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(50);
    digitalWrite(LED_BUILTIN, LOW);
    delay(50);
    n = 0;
  }
}

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial/printf("\nSDK version: %s\n", ESP.getSdkVersion());
  delay(100);   // make sure output has drained
}

void loop()
{
  heartbeat(SKIPCNT);
  dotest();
}
```

Borisvl, 2018-02-02:
I also have experienced this problem. Does anyone have an update on this issue?

Electrozest, 2018-04-15:
I am seeing exact same issue when esp_light_sleep_start() is called. I am typically calling with a delay less than 1000ms set with the esp_sleep_enable_timer_wakeup(microseconds) function. Periodically at random the esp_light_sleep_start() function blocks and never returns from this function.

Even worst, the esp_light_sleep_start() function is enabling and disabling the hardware watchdog timer, and the hardware watchdog timer is sometimes in the disabled state when the block occurs, which means that the CPU effectively stalls with no hardware WDT reset or autonomous recovery from an error state.

Calling the esp_light_sleep_start() function once every second the ESP32 becomes unusable after a few (1-3) hours and a manual reset is then required to recover. This makes the ESP32 unsuitable for tasks requiring remote operation if light sleep is required and hardware WDT is not guaranteed to operate.

Electrozest, 2018-04-15:
This is a critical issue for anyone considering the use of the ESP32 with light sleep mode.

tferrin, 2018-04-16:
Note that Espressif is planning on implementing "automatic" light sleep in the IDF soon, as mentioned here: https://github.com/espressif/arduino-esp32/pull/1165#issuecomment-376717287

closed 2019-08-15 by stale[bot]