espressif/arduino-esp32 #909: WiFiServer receive GET parameters

krmax44, 2017-12-09:
I'm currently doing some testing with the SimpleWiFiServer example sketch. For my project, I need to pass dynamic variables between the client and the server (ESP32).

Let's say, I make the following request: `http://myesp32/test?message=hello`
Now I'd like to get the `GET` parameter `message`. On the ESP8266, you can use `server.arg("message")`, and it would spit out `hello`, but this is not working on the ESP32. How can I get this working?

me-no-dev, 2017-12-11:
that requires parsing the full http stuff :) in the pull requests there is a PR with the 8266 web server. Also you can run ESPAsyncWebServer which works on ESP32

closed 2017-12-11 by me-no-dev

krmax44, 2017-12-11:
ok, i rather get an ESP8266.