espressif/arduino-esp32 #910: ESP32 arduino preferences maximum size (ESP32 dev module)

rammere, 2017-12-09:
### Hardware:
Board:	ESP32 Dev module						?ESP32 Dev Module?
Core Installation/update date:	Last update 11/DEC/2017?
IDE name:						Arduino IDE 1.12
Flash Frequency:	80Mhz
Upload Speed:115200

### Description:
I need to know the maximum size I can store in one name space in the preferences 
and can I change that size ?
thank you.

me-no-dev, 2017-12-11:
under 2K for a single item, but also limited by the size of the nvs partition and everything else stored in there. How much are you trying to store? Use SPIFFS maybe?

rammere, 2017-12-11:
Hi,
Ya thank you, I end up using SPIFFS . is there any downside using SPIFFS I was able to store 1K lines each line is about 30 bytes .
Regards,

me-no-dev, 2017-12-11:
nah SPIFFS is better in this case ;) NVS in it's current config would not fit 30x1K items

closed 2017-12-11 by me-no-dev

dhavals1212, 2022-05-03:
So, what exactly is the maximum size of data I can store, let's say if I erase the entire NVS? Will it be total 20 kB of NVS storage? I have been looking for this answer for long.

dhavals1212, 2022-05-04:
> under 2K for a single item, but also limited by the size of the nvs partition and everything else stored in there. How much are you trying to store? Use SPIFFS maybe?

So, does that mean if I have a nvs storage of 20 kB, I can practiacally create total 10 namespaces worth 2k each? Since, I read somewhere that total of only 16 kB utilised is it safe to assume that if I wipe out data from NVS completely, I may be able to utilise upto 16 kBytes of data? That is managing the namespaces having certain key value pairs? And I am aware that there is limit to String and blob type of data from espressif, although this portion is unclear.