espressif/arduino-esp32 #913: 880µA deep-sleep current Lolin 32 LIte with REV 1 chip

al1fch, 2017-12-10:
### Hardware:
Board:							 Lolin 32 Lite (REV 1 chip)
Core Installation/update date:			24/11/2017
IDE name:					        Arduino IDE 1.8.5
Flash Frequency:					40Mhz
Upload Speed:						115200 bauds

### Description: 

Problem : On Lolin 32Lite  (Rev 1 chip) current is 880µA.... just as light sleep current according to data sheet.  The chip seems to be in light sleep mode, not deep sleep mode

No problemn With  Lolin 32 (REV0 chip).

Same sketch on two boards.
Current is mesasured with same conditions (Lipo input connector ) so the small regulator's current is taken in account.

### Sketch : unmodified code from example :    ESP32/Deep Sleep/Timer Wake up

pipi61, 2017-12-10:
Hi!
In esp32 sleep state the usb-serial adapter not in sleep mode.
On lolin32 and lolin32lite same usb/serial controller?

al1fch, 2017-12-10:
Hi !
-when Lolin32 (normal and Lite) are direct powered from LiPo and usb disconnected (no signal on D+ and D- pins)  usb-serial chips  seems in low power mode. (=suspend mode)
-CP2104 chip on Lolin32, CH340 chip on Lite version

Very low current with D1 mini boards with CH340 , in same configuration  direct 3V3 powered  and usb disconnected. That seems to confirm sleeping (suspend mode) usb-serial chips when D+/D- are unconnected. (less than 80µA in this state according to data sheet)
.... si I don't think that 800µA mentionned here goes to serial usb chip

Problem in sleeping modes for D1 revision of ESP32 ?

closed 2017-12-10 by al1fch