espressif/arduino-esp32 #914: Esp32 multiplexing won't work with arduino library.

OAXP, 2017-12-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
I think that this library won't work with esp32, does anyone have a solution?

### Sketch:
/**
 * This example demonstrates how to write digital signals
 * It assumes there are LEDs+resistors with the positive lead of the LEDs
 * connected to the 16 channels of the 74HC4067 mux/demux, respectively
 * 
 * For more about the interface of the library go to
 * https://github.com/pAIgn10/MUX74HC4067
 */

#include "MUX74HC4067.h"

// Creates a MUX74HC4067 instance
// 1st argument is the Arduino PIN to which the EN pin connects
// 2nd-5th arguments are the Arduino PINs to which the S0-S3 pins connect
MUX74HC4067 mux(13, 12, 14, 15, 27);

void setup()
{
	// Configures how the SIG pin will be interfaced
	// e.g. The SIG pin connects to PIN 3 on the Arduino,
	//      and PIN 3 is a Digital Output
	mux.signalPin(34, OUTPUT, DIGITAL);
}

// Writes to the 16 channels a HIGH value, one after the other
void loop()
{
	for (byte i = 0; i < 2; ++i)
	{
		// Connects to channel i and writes HIGH
		mux.write(i, HIGH);
		delay(25);
	}
	
	mux.disable();  // Disconnects the SIG pin from any channel
	delay(800);
}
### Debug Messages:
Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

C:\Users\OAXP\Documents\Arduino\libraries\MUX74HC4067-master\MUX74HC4067.cpp: In member function 'uint8_t MUX74HC4067::write(int8_t, uint8_t, int8_t)':

C:\Users\OAXP\Documents\Arduino\libraries\MUX74HC4067-master\MUX74HC4067.cpp:138:52: error: 'analogWrite' was not declared in this scope

  if ( type == ANALOG ) analogWrite(signal_pin, data);

                                                    ^

C:\Users\OAXP\Documents\Arduino\libraries\MUX74HC4067-master\MUX74HC4067.cpp:142:61: error: 'analogWrite' was not declared in this scope

   if ( signal_pin_status == 0 ) analogWrite(signal_pin, data);

                                                             ^

exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

OAXP, 2017-12-10:
Library page: http://blog.codebender.cc/2014/01/30/mux74hc4067/

me-no-dev, 2017-12-11:
34 and above are input only pins ;) can not use for output! Lib itself is fine

closed 2017-12-11 by me-no-dev

OAXP, 2017-12-11:
@me-no-dev Ok thanks I'll try with another pin. :)

Makica, 2020-06-12:
It is a library problem, not pin's.

thijstriemstra, 2020-06-27:
> It is a library problem, not pin's.

Agreed.

silveirago, 2023-01-17:
Did anybody have any luck?

mrengineer7777, 2023-01-17:
@silveirago You'd have to ask the library owner https://github.com/pAIgn10/MUX74HC4067.  Not an issue with this repository.