espressif/arduino-esp32 #915: esp_deep_sleep_enable_ext0_wakeup howto de-register the wakeup.

helmut64, 2017-12-10:
I use esp_deep_sleep_enable_ext0_wakeup() to register GPIO pins for wakeup during light sleep and deep sleep. Once it is registered and an wakeup occurs I like to de-register this port before going to sleep again.

Any ideas how this can be done?

me-no-dev, 2017-12-11:
you want to sleep for particular time after you wake up by GPIO?

helmut64, 2017-12-11:
No the wakeup works, I go to sleep again, and it wakes up me again and again because the esp_deep_sleep_enable_ext0_wakeup is still registered. 

Basically a button e.g. BOOT is pressed, once I receive a wakeup I like to de-register the pin-0 wakeup and go to sleep again.

At present I receive about 10 wakeups because the button is bouncing and I cannot de-active the wakeup on the pin.

me-no-dev, 2017-12-11:
maybe detect that wakeup and do not call esp_deep_sleep_enable_ext0_wakeup again? can you post a minimal sketch? You can debounce by adding a cap to ground and pull-up resistor (100nF should do)

helmut64, 2017-12-11:
Lets say it can be also an external device where the pin stays low and I need to go to sleep for saving energy, but I don’t like to get the wakeup again.

To register pin 0 for high it do:
esp_deep_sleep_enable_ext0_wakeup((gpio_num_t)0,1)
To register pin 0 for low I do:
esp_deep_sleep_enable_ext0_wakeup((gpio_num_t)0,0)

However I have not found a way to specify disable this wakeup pin, once I started registering it with esp_deep_sleep_enable_ext0_wakeup() i cannot de-register it.

I will do a little sample app.

helmut64, 2017-12-11:
[ESP32_sleeptest.zip](https://github.com/espressif/arduino-esp32/files/1549244/ESP32_sleeptest.zip)
Here a little sample project, using light sleep (some occurs using deep sleep)
When the "#if 1" is disabled I receive the correct behaviour, when it is enabled I received about 20 wakeups and often the Interrupt routing gets called multiple times.

I believe it is related to esp_deep_sleep_enable_ext0_wakeup((gpio_num_t)0,1) which does not allow to de-register the wakeup.

I use the ESP32  DevKit-C with Rev1.
Thank you for investigating into it.

closed 2019-08-12 by helmut64