espressif/arduino-esp32 #917: Ethernet demo compiles but - Windows works / Mac(OS X) fails

gjt211, 2017-12-11:
### Hardware:
Board:							Olimex ESP32 Gateway
Core Installation/update date:			10/12/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
#### Source file `ETH_LAN8720.ino` demo
I complied the `ETH_LAN8720` demo initially on my Windows7 laptop, flashed the OlimexESP32 Gateway and it worked fine.
I then tried the exact same demo at home on my Mac. I flashed it on the same board. It only shows `ETH started` and then `ETH Connected` in the terminal but never gets any further showing an IP address etc.
I then returned the board back to the Windows PC, flashed the exact same demo program and it works fine again. I tried updating the core files/idf etc as per the guides and there is no change.

On the Mac, if I disconnect or reconnect the Ethernet cable the serial terminal shows `ETH Disconnected` and `ETH Connected` so it is definitely seeing the Ethernet cable and switch.

Nothing has been changed (or needs to be changed) for this demo file, so I am lost as to why this works on my windows PC, but not on my Mac.

### Sketch:
```cpp

/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

gjt211, 2017-12-11:
#### Update
Not sure if it is relevant, but when compiling on my Mac PC, I get a lot of the following warnings
````
<command-line>:0:14: warning: ISO C++11 requires whitespace after the macro name
````
and
````
<command-line>:0:14: warning: ISO C99 requires whitespace after the macro name
````

me-no-dev, 2017-12-11:
are you sure you are running the latest code on your mac? check if you are getting ```SYSTEM_EVENT_STA_GOT_IP ```. It was messed up a few commits back, but was since fixed

gjt211, 2017-12-11:
@me-no-dev, thanks for your comments. Will double check, I am sure I did run the code to update the core.
As another test, I just got out my MacBookPro and tried the same example file on that. I still get the warnings (same as my main Mac PC), but the code is working, so there must be some problem with my main Mac PC. Will keep looking into it and report my findings.

gjt211, 2017-12-11:
@me-no-dev 
I just deleted the esp32 folder and re-run...
```
mkdir -p ~/Documents/Arduino/hardware/espressif && \
cd ~/Documents/Arduino/hardware/espressif && \
git clone https://github.com/espressif/arduino-esp32.git esp32 && \
cd esp32 && \
git submodule update --init --recursive && \
cd tools && \
python get.py
```
and all is working well.

My actual Arduino folder is located elsewhere (not shown above) but all is good.
Sorry for the botherI may have caused, and thanks again.

closed 2017-12-11 by gjt211

me-no-dev, 2017-12-11:
glad it worked :)