espressif/arduino-esp32 #918: AUTH_FAIL loop following disconnect until ESP32 is power cycled.

lonerzzz, 2017-12-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			2017/11/26
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Not sure if this issue already exists here or in the ESP32-IDF issues, but didn't see it.

After one of the recent core updates, I have noticed that in the event that the WiFi gets disconnected or the application calls WiFi.disconnect(true|false) such as when a WiFi.scan() has to be called (due to a known bug https://github.com/espressif/arduino-esp32/issues/758), that it is often no longer possible to reconnect to the WiFi router until the ESP32 is power cycled. The problem happens about 1 in 3 times. Sometimes a simple reboot fixes the problem, but generally the power cycle is required.

Every attempt to make a connection with WiFi.begin results in AUTH_FAIL in this scenario despite the WiFi.begin parameters being correct:

[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 201 - AUTH_FAIL

The following message sometimes occurs as well, but not always so it is not clear :
[W][WiFiGeneric.cpp:270] _eventCallback(): Reason: 8 - ASSOC_LEAVE

ctruter, 2018-02-01:
@lonerzzz I have a simmilar issue, perhaps try to use a version _before_ the 2017/11/24 update ([81a9c45](https://github.com/espressif/arduino-esp32/commit/81a9c45a1ef5833e80c9235ff393a7641c669a94)) - that worked for me.

I think the update of the IDF to e5b2c1c might have caused this behavior with the WiFi

lonerzzz, 2018-02-12:
@ctruter Thanks for the suggestion. Unfortunately, I need other functionality later than that date so I am stuck on this one.

projectgus, 2018-06-26:
@lonerzzz I think this issue was addressed earlier in the year in IDF. Are you still seeing any problems like this with recent Arduino versions?

chegewara, 2019-01-25:
Hi @projectgus , its long time ago closed issue, but im having similar problem now with v1.0.1.
What is very strange all has been working for quite some time, but after teens of flashing during development process i have `ASSOC_LEAVE` and other errors during connect using WiFiMulti.

Another weird thing is that after changing esp32 all is working now (yes, ive been doing make erase_flash, but it wont help).

stickbreaker, 2019-01-25:
@chegewara compare the NVS data from the non working one to your new working one. In my github i have a nvs display repo.

Chuck.

projectgus, 2019-01-28:
@chegewara What AP are you connecting to? Is it possible to test with a different AP (maybe set up a phone hotspot or something) and see if the problem only happens with a certain AP vendor or some other hardware variable?

chegewara, 2019-01-29:
@projectgus I am using smartphone AP, from samsung S9+. I dont have any issue with this AP and esp32, only problem is with that particular esp32 board, but this board worked earlier with the same code. 
I flashed other esp32 board and all is ok.

Maybe i flashed that one board too many times and flash/nvs is faulty now.

closed 2019-08-15 by stale[bot]