espressif/arduino-esp32 #919: ESP32 disconnecting from system port after downloading wifi example

Arun2102, 2017-12-12:
### Hardware: ESP32
Board:	EPS32 WROOM-32			ESP32 Dev Module
Core Installation/update date:		
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
My esp32 is woking fine for blink example program but when i download SimpleWiFiServer example, the esp32 is getting disconnected from my system port. in this situation i couldnt connect serial port. Also the esp32 is ot connected to the specfied ssid network.
There is no errors during compilation. i have ony entered the ssis and password of my network in the example. nothing else is altered.
The problem occurs for all the wifi example codes. the board is working fine with other examples.

lonerzzz, 2017-12-14:
@Arun2102 Are you holding down the proper 2 pins to set it into bootloader mode and reset? Have you been able to program successfully?

Arun2102, 2017-12-14:
Thank you for your reply Jason. I can able to program successfully. But
after programming the simpleWiFiserver code I couldn't program any other
code.
To program some other example I just pressed and hold down the reset and
when the ide shows connecting il release the reset so that my code will be
programmed successfully. This condition occurs only when I have downloaded
the WiFi examples.
And Janson what is that 2pins to set into bootloader? I couldn't get about
it.

PRO2XY, 2017-12-20:
Sounds like a power supply issue. Are you powering the module using the 3.3V from the USB-Serial converter? If yes, that's your issue. You must use a power source capable of delivering at least 500mA at 3.3V to the module. Also, don't forget, wire length and solder joint quality also affect the actual power delivered.

Arun2102, 2017-12-22:
Thanks for your suggestion.
If i power the module with a separate supply then how i can use serial monitor? 
Also this problem occurs only for the wifi codings. rest all example codes works fine with the USB supply itself.

PRO2XY, 2018-03-20:
I don't know if you got around this issue (it's been > 3 months :-p ), but the reason the problem occurs with WiFi sketches is because WiFi is power intensive. 
As for powering with a separate supply, you connect the ground of the power supply and your serial monitor together. So, if you're using a NodeMCU type board, you can use the USB port for communication, and attach external power supply on the 5V (or 3.3V, if your power supply is 3.3V) pins.

kowsche, 2018-03-27:
I had the exact same problem with COM port getting disconnected when trying out the WiFi samples. The problem was fixed when I used a better quality USB cable to connect from my Laptop to the ESP32 board!!

closed 2018-04-19 by everslick