espressif/arduino-esp32 #920: arduino ide iBeacon example compile

qrpfun, 2017-12-12:
### Hardware:
Board:			    				       ESP-WROOM-32
Core Installation/update date:			10 December 2017 
IDE name:							Arduino IDE 
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:
I tried to compile the esp-idf example ble_ibeacon (https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_ibeacon) under arduino ide but it's not so easy to understand how to recreate the right environment for the task.
Is there any best practice to migrate an idf example to arduino-ide or any other suggestion ?
thanks in advance
kind regards
Giampiero

chris-gunawardena, 2017-12-18:
Have you looked at this one? https://github.com/espressif/arduino-esp32/blob/master/libraries/SimpleBLE/examples/SimpleBleDevice/SimpleBleDevice.ino

If you want a quick hacky beacon to send data you can append data to the end of the beacon id. `ble.begin("beacon1: tmp: " + int_temp_reading);`

qrpfun, 2017-12-18:
tanks Chris. I looked at that example too but is not possible to fill the right data structure.
Now I'm looking in this direction: https://github.com/nkolban/esp32-snippets/issues/271

copercini, 2018-04-21:
Closed due it was solved on https://github.com/nkolban/esp32-snippets/issues/271 =)

closed 2018-04-21 by copercini