espressif/arduino-esp32 #921: "Guru Meditation Error" connecting to WPA2-Enterprise

SijmenSchoon, 2017-12-12:
### Hardware:
Board:						NodeMCU-32S
Core Installation/update date:	79e5d4c
IDE name:					Arduino IDE
Flash Frequency:				80 MHz
Upload Speed:				921600

### Description:
When trying to connect to eduroam (a WPA2-Enterprise network), the ESP32 crashes with a Guru Meditation Error while (presumably) checking the certificate:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4015624d  PS      : 0x00060930  A0      : 0x80156fb4  A1      : 0x3ffdc600  
A2      : 0x3ffe3fb4  A3      : 0x00000001  A4      : 0x3ffe1ddc  A5      : 0x3ffe1ddb  
A6      : 0x00000002  A7      : 0x00000005  A8      : 0x8015624b  A9      : 0x3ffdc5e0  
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x3ffe3fb4  A13     : 0x00000000  
A14     : 0x00414320  A15     : 0x00000010  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4015624d:0x3ffdc600 0x40156fb1:0x3ffdc620 0x4015252a:0x3ffdc640 0x400e5132:0x3ffdc670 0x400e531b:0x3ffdc720 0x400e207e:0x3ffdc7f0 0x400e2481:0x3ffdc830 0x400e16dd:0x3ffdc8b0 0x400e1140:0x3ffdc900 0x400e1199:0x3ffdc940 0x400df809:0x3ffdc960 0x400e0bd8:0x3ffdc9a0 0x400dd4e9:0x3ffdca00 0x400dd879:0x3ffdca40

Rebooting...
ets Jun  8 2016 00:22:57
```

Decoding the backtrace:
```
0x4015624d: mp_read_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/libtommath.h line 990
0x40156fb1: bignum_set_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/bignum.c line 114
0x4015252a: crypto_mod_exp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c line 40
0x400e5132: x509_certificate_check_signature at ?? line ?
0x400e531b: x509_certificate_chain_validate at ?? line ?
0x400e207e: tlsv1_client_set_cred at ?? line ?
0x400e2481: tlsv1_client_process_handshake at ?? line ?
0x400e16dd: tlsv1_client_handshake at ?? line ?
0x400e1140: tls_connection_handshake2 at ?? line ?
0x400e1199: tls_connection_handshake at ?? line ?
0x400df809: eap_peer_tls_process_helper at ?? line ?
0x400e0bd8: eap_peer_tls_register at ?? line ?
0x400dd4e9: eap_sm_process_request at ?? line ?
0x400dd879: wpa2Task at ?? line ?
```

Commenting out the `esp_wifi_sta_wpa2_ent_set_ca_cert` call does not change the error.

The espressif/esp-idf branch has however done a commit a while ago with "wifi fixes". Maybe that commit fixes the problem?

### Sketch:
Relevant code:

```cpp
#define ESP_ERROR_RETURN(tag, x) { \
        if ((x) != ESP_OK) {       \
            Serial.println(tag);   \
            return false;          \
        }                          \
    }

bool WiFiTool::_internal_connect(const String ssid, const String username,
                                 const String password)
{
    char *mut_username = new char[username.length() + 1];
    strcpy(mut_username, username.c_str());

    char *mut_password = new char[password.length() + 1];
    strcpy(mut_password, password.c_str());

    Serial.print("USERNAME: ");
    Serial.println(mut_username);
    Serial.print("PASSWORD: ");
    Serial.println(mut_password);

    // Set the WPA2 enterprise CA certificate
    ESP_ERROR_RETURN("set_ca_cert", esp_wifi_sta_wpa2_ent_set_ca_cert(
        (uint8_t *)Certificates::digicert, strlen(Certificates::digicert)));

    // Set identity, username and password
    ESP_ERROR_RETURN("set_identity", esp_wifi_sta_wpa2_ent_set_identity(
        (uint8_t *)mut_username, strlen(mut_username)))

    ESP_ERROR_RETURN("set_username", esp_wifi_sta_wpa2_ent_set_username(
        (uint8_t *)mut_username, strlen(mut_username)));

    ESP_ERROR_RETURN("set_password", esp_wifi_sta_wpa2_ent_set_password(
        (uint8_t *)mut_password, strlen(mut_password)));

    // Connect to the network
    esp_wpa2_config_t wpa2_config = WPA2_CONFIG_INIT_DEFAULT();
    ESP_ERROR_RETURN("enable", esp_wifi_sta_wpa2_ent_enable(&wpa2_config));

    WiFi.begin(ssid.c_str());
    return true;
}

```

### Debug Messages:
That option does not seem to be there.

allertman, 2018-02-27:
After I upgraded to the latest Arduino IDE (1.8.5) and ESP32 core (1.22.0-80-g6c4433a-5.2.0) yesterday (26 Feb 2018), I experienced the same issues as described by Sijmen. Does anyone have a solution?

Redferne, 2018-03-02:
I can confirm this, here is a dump:
```
ESP SDK: v3.1-dev-239-g1c3dd23f-dirty
Flash Size: 4194304
Flash Speed: 40000000
WIFI Enterprise...Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x401aceb1  PS      : 0x00060d30  A0      : 0x801adc18  A1      : 0x3ffe62d0  
A2      : 0x3ffea3d8  A3      : 0x00000001  A4      : 0x3ffe9180  A5      : 0x3ffe917f  
A6      : 0x00000002  A7      : 0x00000005  A8      : 0x801aceaf  A9      : 0x3ffe62b0  
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x3ffea3d8  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000010  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x401aceb1:0x3ffe62d0 0x401adc15:0x3ffe62f0 0x401a916a:0x3ffe6310 0x401373ce:0x3ffe6340 0x401375cf:0x3ffe63f0 0x40134126:0x3ffe64c0 0x4013453d:0x3ffe6500 0x4013372d:0x3ffe6580 0x4013315c:0x3ffe65d0 0x401331c1:0x3ffe6610 0x401316b2:0x3ffe6630 0x40130dbf:0x3ffe6670 0x4012f1ed:0x3ffe66d0 0x4012f589:0x3ffe6710

0x401aceb1: mp_read_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/libtommath.h:990
0x401adc15: bignum_set_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/bignum.c:114
0x401a916a: crypto_mod_exp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c:40
0x401373ce: x509_certificate_check_signature at ??:?
0x401375cf: x509_certificate_chain_validate at ??:?
0x40134126: tlsv1_client_set_cred at ??:?
0x4013453d: tlsv1_client_process_handshake at ??:?
0x4013372d: tlsv1_client_handshake at ??:?
0x4013315c: tls_connection_handshake2 at ??:?
0x401331c1: tls_connection_handshake at ??:?
0x401316b2: eap_peer_tls_process_helper at ??:?
0x40130dbf: wifi_sta_get_enterprise_disable_time_check at ??:?
0x4012f1ed: eap_sm_process_request at ??:?
0x4012f589: wpa2Task at ??:?
```

asergios, 2018-04-08:
Any update on this?

GamalBorhan, 2018-04-22:
I have exactly the same issue, I have tried many things but still not working out :(

SijmenSchoon, 2019-08-01:
As far as I know this is still broken

>

lbernstone, 2019-08-01:
There is an example specifically designed and tested for eduroam sites. Have you tested https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples/WiFiClientEnterprise ?

allertman, 2019-08-19:
I just fixed this by including WiFi.mode(WIFI_STA); before the "enable". Goodluck all!

closed 2019-11-01 by stale[bot]