espressif/arduino-esp32 #922: The Watchdog got triggered. (!?)

euquiq, 2017-12-12:
I got here an ESP32 performing some sensor tasks, which was left alone, working for two or three days.  Some minutes ago, I tried to refresh the webpage that the ESP32 serves and it wasn't working anymore.

I did open the COM in order to read the Serial output, just in case, and I got:

Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: loopTask

being periodically  sent at about 5 seconds interval.

The Arduino code is running inside the ESP32, since I got a gas sensor and whenever I trigger it, the Arduino code enables a Relay which I can hear / see turning on ... but the ESP32 has lost wifi connection and it is not returning (Although my code watches for WiFi.status() and if disconnected tries to reconnect).

This may probably be a bug, even at IDF level. I am sorry I do not know what made it trigger, nor have a way of posting any more data. 

I got the latest Arduino framework in use.

I can implement whatever mechanism you may propose in order to catch it better the next time this happens,  If it does.

Regards,

Enrique.

lonerzzz, 2017-12-13:
This suggests you have one or two loops that are not servicing the watchdog by calling delay() periodically with some millisecond value. Since the ESP32 is on a real time OS, it is monitoring for this sort of thing and your threads needs to share their time.

euquiq, 2017-12-14:
It just happened again ...

lonerzzz, 2017-12-15:
@euquiq Sometimes activities in one task/thread can halt activity in the other. It is not a matter of you adding delay on a core that you are using, but you have to understand any calls that you make which might cause activity on the other core. On the ESP32, that could a troublesome wireless call for example. What you want to determine is if you have long running, CPU intensive activities within your code. You could, for example, have  the loop calling delay() regularly but also have a method in that loop that monopolizes the processor for some period of time long enough to trigger the watchdog. In that case, you would need a delay within the method that you are calling. Hope that helps.

euquiq, 2017-12-19:
@lonerzzz Thanks again for your explanation. I am closely inspecting my code. I am closing this issue, since as you comment, it surely is some exotic situation from my wifi network / my code which incurs into such behavior on core 0. If I get to clarify something worth mentioning, of course I will report back so it may help others. Regards, Enrique.

closed 2017-12-19 by euquiq

akshar001, 2018-08-17:
Anyone who still want to avoid that error should try
adding these three line to your continuous  rtos function, 

`TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;`
 `TIMERG0.wdt_feed=1;`
 `TIMERG0.wdt_wprotect=0;`

And add this header
```
#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
```

1technophile, 2018-10-01:
@akshar001 Hello,

Thanks for the tip, nevertheless in the scope of the arduino-esp32 environment what do you mean by continuous rtos function, are we talking about the main loop of the arduino program or the task affected to core 0 in my case?

https://community.openmqttgateway.com/t/ble-and-esp32-dev-environment/102/20?u=1technophile

Regards

akshar001, 2018-10-15:
@1technophile what i meant by any rtos task which runs without a delay,loop is actually a rtos task which is implemented in esp32 arduino so by just putting it into a loop can solve your problem.(there is no problem of core)

1technophile, 2018-10-15:
@akshar001, thanks for the info we will test your solution

capedra, 2018-10-17:
> @1technophile what i meant by any rtos task which runs without a delay,loop is actually a rtos task which is implemented in esp32 arduino so by just putting it into a loop can solve your problem.(there is no problem of core)

@akshar001 @me-no-dev I'm still getting the following warning:
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: handleSecondCore
CPU 1: loopTask
```
where should I put these lines of code that you provided? Thanks!

capedra, 2018-10-17:
@me-no-dev @1technophile @akshar001 
I changed my code in a way that I don't need to write ```**while (true)**``` or ```**for (;;)**``` inside the core 0 function, and, this way, I just need to put **delay(1)** inside the method that handles core 0 loop. This way, I don't need these three lines of code that @akshar001 provided. This is how I did it:
```
void feedGPS()
{
  while (SerialGPS.available() >= 0)
  {
    if (SerialGPS.available() == 0)
    {
      //Serial.println(F("D"));
      //vTaskDelay(10); // wrapper for delay()
      delay(1);
    }
    else
    {
      char c = SerialGPS.read();
      GPSobj.encode(c);
      //Serial.write(c);
    }
  }
}
```

wan420, 2018-11-08:
> 
> 
> > @1technophile what i meant by any rtos task which runs without a delay,loop is actually a rtos task which is implemented in esp32 arduino so by just putting it into a loop can solve your problem.(there is no problem of core)
> 
> @me-no-dev @1technophile @akshar001
> UPDATE 1: The warning disappear when I put these three lines of code inside **`void loop() {}`**. I'm currently testing if the board will reset or shutdown anytime soon. The ESP-32 board that I'm currently using is the LOLIN D32 Pro. Thanks!

This was 22days ago, can you please share the result of the test, I am using nodemcu 32s and currently experiencing watchdog reset, I used @1technophile's solution and recently uploaded the code to esp32s. I would like to know before hand as to what happend with your tests. Ty

akshar001, 2018-11-22:
> This was 22days ago, can you please share the result of the test, I am using nodemcu 32s and currently experiencing watchdog reset, I used @1technophile's solution and recently uploaded the code to [...]

Anyone who still want to avoid that error should try
adding these three line to your continuous rtos function,

TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;

And add this header

#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"

ucar19, 2020-06-23:
> Anyone who still want to avoid that error should try
> adding these three line to your continuous rtos function,
>
> `TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;`
> `TIMERG0.wdt_feed=1;` [...]

I wrote what  you write and I create infinitive loops in one task. Task Watchdog didn't trigger. So If I write your code, the watch dog never triggered. It is not good I think.

Eiyvor, 2020-11-27:
disabling the TWDT is not a solution. Doing so you may not get the watchdog triggered error message and your ESP wouldn't restart but your lower priority tasks will still starve for CPU time and things wouldn't work the way you want it to.

akshar001, 2020-11-29:
@Eiyvor in most of the scenario i am running till now, it works well. You are right by the way it is not a solution. but recent workaround. because problem is from esp inner firmware and we do not have full source code of them

akshar001, 2020-11-29:
> I wrote what you write and I create infinitive loops in one task. Task Watchdog didn't trigger. So If I write your code, the watch dog never triggered. It is not good I think.

Can you share me a code here. It depends on how you put it

mduchain, 2023-08-04:
I found this thread as I was facing the same issue.
I had random reset due to watchdog timer on wifi task - cpu 0, generally every 10 to 15 minutes.

In my case, I had reduced the CPU clock frequency to 40Mhz (from 240) to lower battery consumption.
Raising the frequency to 80Mhz resolved this behavior.

mortezaMoezzifar, 2024-05-05:
If we reach a loop like the one below in the program, it causes the ESP32 to reset:
`while (true) {
    // nothing
}`

mathieucarbou, 2024-05-05:
> If we reach a loop like the one below in the program, it causes the ESP32 to reset: `while (true) { // nothing }`

that is not a bug and that is what I would expect from the WDT, which is here to _"detect instances of tasks running for a prolonged period of time wihtout yielding. This can be an indicator of poorly written code that spinloops on a peripheral, or a task that is stuck in an infinite loop."_, quoting [the doc](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/wdts.html)

Try add some `yield()` or short `delay()` calls or, ideally, remove this loop and refactor your code to use a state machine.