espressif/arduino-esp32 #923: Suport for psram

Matheus-Garbelini, 2017-12-12:
Hello, are there any plans to support psram when compiling code?
I tried to call init_psram after including some headers but the linker wasnt able to find any code to initialize psram. Thanks

HelloDB, 2018-01-11:
I need it，too

Matheus-Garbelini, 2018-01-11:
@HelloDB I'm using EspIDF with Arduino as component and have to say that everything is working correctly. And have to say that ESP32 is the best thing ever invented. Hahaha, the ST microelectronics with its CORTEX M7 stands no chance in terms of cost-benefit.

manecolooper, 2018-03-07:
@Matheus-Garbelini  , is psram also working under arduino environment? thanks!

Matheus-Garbelini, 2018-03-07:
@manecolooper you can flash esp and it works normally, but at the time the esp32 didn't initialize the spram as it can when using idf.
The solution i got was to use IDF with arduino component so i could enable psram support on initialization.

manecolooper, 2018-03-07:
Thanks, i'll try to get it working as component in idf, my tests so far give compilation errors even with the simplest sketchs, but of course i'm doing something wrong, best regards!!!

copercini, 2018-07-01:
It's supporting PSRAM now =)
https://github.com/espressif/arduino-esp32/commit/184bb069f2b816c897d5151307d8bf0d6519895a

closed 2018-07-01 by copercini

Matheus-Garbelini, 2018-07-16:
@copercini thank you.

copercini, 2018-07-16:
@Matheus-Garbelini you should thank @me-no-dev not me =P