espressif/arduino-esp32 #925: Follow MacOS install procedure, but board doesn't appears in Tools>Board menu

christophepersoz, 2017-12-13:
Hello,

I followed the commands lines that are indicated on this page :
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md

All goes nicely, no errors, but even after a fresh restart of Arduino IDE, no ESP-32 board is mentioned into the Tools > Board menu list.

Anyone got the same issue or have a workaround to make it works ?
Thanks in advance!

qrpfun, 2017-12-14:
which macos are you using ? did you have any error during installation process ? I had followed the described procedure on two different mac and everything works fine

christophepersoz, 2017-12-14:
I did use the ones of the pages mentioned in my post.
What is weird is that the folder created `~/Documents/Arduino/hardware/espressif` seems not common for hardware definition on MacOS with Arduino IDE.

Usually the hardware definition are taking place either in the folder `~/Library/Arduino15/packages/hardware/` or in the Arduino application package like Teensy board does.

Here the result of the install as mentioned in the link above :
`Cloning into 'esp32'...
remote: Counting objects: 6840, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 6840 (delta 0), reused 1 (delta 0), pack-reused 6832
Receiving objects: 100% (6840/6840), 103.62 MiB | 6.92 MiB/s, done.
Resolving deltas: 100% (3879/3879), done.
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into '~/Documents/Arduino/hardware/espressif/esp32/libraries/BLE'...
Submodule path 'libraries/BLE': checked out 'f8fe9d7cdfb20caa54b70849826d1ac6e375ff78'
System: Darwin, Info: Darwin-16.7.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Downloading xtensa-esp32-elf-osx-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-osx-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Downloading esptool-4dab24e-macos.tar.gz
Done
Extracting esptool-4dab24e-macos.tar.gz
Downloading mkspiffs-0.2.1-osx.tar.gz
Done
Extracting mkspiffs-0.2.1-osx.tar.gz
Renaming mkspiffs-0.2.1-osx to mkspiffs
Done`

And even after the third install, I still do not have the board appears inside the Board list, where I already have Teensy, SAMD M0+, Feather nRF52, RedbearLab Nano, and AVR. Really strange, but not working at all on my config.

qrpfun, 2017-12-15:
I can confirm you there is no special issue related to the folder used.
Everything here works fine under: ~/Documents/Arduino/hardware/espressif/esp32
Which are macos and arduino ide release ?
Here I'm using latest IDE 1.8.5 and Sierra 10.12.6

christophepersoz, 2017-12-15:
I'm using MacOS 10.12.6 and Arduino IDE 1.8.5, same as yours...
Does the preference.txt file can interfere with this ? Just guessing...

qrpfun, 2017-12-15:
mhh .. very strange
I don not think
do you have sketchbook.path configured to your ~Documents/Arduino path ?
did you run the python script at installation stage ?
If nothing work may ba you can restart installing arduino from scratch
changing original folders names

christophepersoz, 2017-12-15:
Yes, my sketchbook path is : ~/Developer/_Arduino
I just ran the all the command in terminal as mentioned in the first link.

qrpfun, 2017-12-15:
> Yes, my sketchbook path is : *~/Developer/_Arduino*

 is "*_Arduino" with underscore _* a typing error  ?

christophepersoz, 2017-12-15:
nope. Is that the issue ?

qrpfun, 2017-12-15:
I see in you very first msg an other folder, the right one:
~/Documents/Arduino/hardware/espressif

qrpfun, 2017-12-15:
> That was the commands entered in the Terminal, as mentioned in the link
> https://github.com/espressif/arduino-esp32/blob/master/
> docs/arduino-ide/mac.md
>
> I didn't know that I have to change them to make everything working. I
> have many other boards, and that the only one which cause a problem.

ok... so now it works ?

christophepersoz, 2017-12-15:
Yes!!! It could be nice to indicate somewhere in the documentation that if the path are not the original ones you need to change them accordingly. Thanks a lot.

closed 2017-12-15 by christophepersoz

qrpfun, 2017-12-15:
ok but for my installations those folders are the originals one .. never
seen _Arduino ...
Anyway you are welcome

ciao

beegee-tokyo, 2017-12-15:
I started to work on a Wiki that hopefully will be merged into the main report sometime. [arduino-esp32 Wiki](https://github.com/beegee-tokyo/arduino-esp32/wiki).
Will add an entry about this tomorrow.

nobeans, 2018-01-04:
I faced the same problem.
My sketch book had been also customized to "~/Developer/something".
After I've read this issue and changed to "~/Decuments/Arduino", it works and the "ESP32 Arduino" menu was finally revealed in menu!

But I couldn't understand why the setting is related. It seems not good. I think the additional board settings shouldn't be dependent on the directory path of "sketch book's default path".