espressif/arduino-esp32 #926: wifi.mode(WIFI_AP_STA)  error

auladinamica, 2017-12-13:
WiFi.mode(WIFI_AP_STA);

error :
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 13 - AP_STOP

whats is wrong ?

thanks

lonerzzz, 2017-12-14:
@auladinamica Please provide a code example indicating what you are calling in more detail. Are you subsequently calling something like: WiFi.begin( SSID, wifiPresharedKey ); ?

Are you intentionally setting up both an access point and infrastructure or do you only want one?

auladinamica, 2017-12-14:
Thanks @lonerzzz  i want conect to my wifi in channel 1 , and enable espnow to receive data via channel 1 too.  Here is the code !!!...

If you compile  and disable call to checkConection() function , the program receive data... but also the debug send a error !!!! 

/*
 if( !checkConnection()){

    ESP.restart();
  }
*/
-------------------------------------------------------------------------------------------------------------------this is the debug output .

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
ESPNow/Basic/Slave Example
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP
AP Config Success. Broadcasting with AP: OMMA_Slave_30:AE:A4:04:F9:10
AP MAC: 30:AE:A4:04:F9:11[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP

ESPNow Init Success
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 7 - STA_GOT_IP

if you reset again the esp32,  this is the debug output..

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
ESPNow/Basic/Slave Example
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP
AP Config Success. Broadcasting with AP: OMMA_Slave_30:AE:A4:04:F9:10
AP MAC: 30:AE:A4:04:F9:11
ESPNow Init Success[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP
[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 13 - AP_STOP

[D][WiFiGeneric.cpp:266] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:271] _eventCallback(): Reason: 201 - AUTH_FAIL
Last Packet Recv from: 30:ae:a4:21:9a:80
Last Packet Recv Data: 1
largo data           : 20
-----------------------------------------------------------------------
Id    Sensor     :1
Tipo Sensor      :1
Temp Aire        :26.00
Hume Aire        :50.00
Hume Tierra      :519
-----------------------------------------------------------------------
Last Packet Recv from: 30:ae:a4:21:9a:80
Last Packet Recv Data: 1
largo data           : 20
-----------------------------------------------------------------------
Id    Sensor     :1
Tipo Sensor      :1
[... 102 lines cut ...]
  snprintf(macStr, sizeof(macStr), "%02x:%02x:%02x:%02x:%02x:%02x",
           mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
  Serial.print("Last Packet Recv from: "); Serial.println(macStr);
  Serial.print("Last Packet Recv Data: "); Serial.println(*data);
  Serial.print("largo data           : "); Serial.println(String(data_len));
    
    
    memcpy(&sensorData_Higrow, data, sizeof(sensorData_Higrow));
    
    if (sensorData_Higrow.tipo_sensor==1){  
      Serial.println("-----------------------------------------------------------------------");
      Serial.print("Id    Sensor     :" ); Serial.println(String(sensorData_Higrow.id_sensor));
      Serial.print("Tipo Sensor      :" ); Serial.println(String(sensorData_Higrow.tipo_sensor));
      Serial.print("Temp Aire        :" ); Serial.println(String(sensorData_Higrow.temp_aire));
      Serial.print("Hume Aire        :" ); Serial.println(String(sensorData_Higrow.hum_aire));
      Serial.print("Hume Tierra      :" ); Serial.println(String(sensorData_Higrow.hum_tierra));
      Serial.println("-----------------------------------------------------------------------");
    }
    
   hume[sensorData_Higrow.id_sensor]=sensorData_Higrow.hum_tierra;
}

void loop() {
              
}

copercini, 2017-12-14:
@auladinamica try with `WiFi.mode(WIFI_MODE_APSTA);`   =)

auladinamica, 2017-12-15:
@copercini 

The debug say all is good but esp_now fails, not receive data,  never !!! 

### I can`t conect wifi sta_mode and ap_mode to receive esp_now data  at the same time !!!!

why ? what i do wrong ?

 
------------------------------------------------------------------------------
void ReconectarMqtt() {
  
    int aux=0;
    char topic1[20];
   
   String top="temp-";
   String id=idchip();
   String topic=top + id;
   topic.toCharArray(topic1, topic.length() +1);
   
 while (!client.connected()) {
 Serial.print("Attempting MQTT connection...");
 aux=aux+1;
 if (client.connect(topic1)) {
  Serial.println("connectado a mqtt");
  client.subscribe("omma-3"); // me suscribi al sensor de humedad       
  client.subscribe(topic1);
 } else {
  delay(500);
  if (aux==5){  
      ESP.restart();
     }
  }
 }
}

-------------------------------------------------------------------------------
void configDeviceAP() {
  String Prefix = "Slave_";
  String Mac = WiFi.macAddress();
  String SSID = Prefix + Mac;
  String Password = "123456789";
  bool result = WiFi.softAP(SSID.c_str(), Password.c_str(), 1, 0);
  if (!result) {
    Serial.println("AP Config failed.");
  } else {
    Serial.println("AP Config Success. Broadcasting with AP: " + String(SSID));
  }
}

 --------------------------------------------------------------------------------
 boolean Leo_conf_eprom() {
  preferences.begin("Config", true);
  wifi_ssid = preferences.getString("WIFI_SSID");
  wifi_password = preferences.getString("WIFI_PASSWD");
  preferences.end();
  

  if( (wifi_ssid.length() > 0) and (idchip().length()>0) ) {
       return true;
} else {
    return false;
  }
}
-------------------------------------------------------------------------------------
void setup() {
  
  Serial.begin(115200);
  Leo_conf_eprom();   // read parameters wifi_ssid and wifi_password.
  delay(50);   

         WiFi.mode(WIFI_MODE_APSTA);
         WiFi.begin(wifi_ssid.c_str(), wifi_password.c_str());
         
         while (WiFi.status() != WL_CONNECTED) {
           delay(1000);
           Serial.print(".");
           }
          
           delay(1000);
          configDeviceAP();
          Serial.print("AP MAC: "); Serial.println(WiFi.softAPmacAddress());
  
          debug("Conectar Mqtt");
          
          client.setServer(mqttServer, mqttPort);
          client.setCallback(callback_Mqtt);
          
          ReconectarMqtt();          
          
        }

        InitESPNow();
        esp_now_register_recv_cb(OnDataRecv);
}
-----------------------------------------------------------------------------------------------

### The out put to debug..

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
M5Stack initializing...OK
....AP Config Success. Broadcasting with AP: OMMA_Slave_30:AE:A4:04:F9:10
AP MAC: 30:AE:A4:04:F9:11
1:6420:Conectar Mqtt
Attempting MQTT connection...connectado a mqtt
ESPNow Init Success

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

@auladinamica when providing code snippets, please use the correct MARKDOWN syntax: three ticks before and after the source code.

closed 2018-04-19 by everslick