espressif/arduino-esp32 #927: after reinstall the core.. sketch does not compile anymore :-(

Buffalchill, 2017-12-14:
I had to reinstall the ESP32 core, now i get compiler errors when i compile my sketch. Other sketches can be compiled well. The sketch was definitly compilable bevor i got 8 devices which are running with the code. 
can you please help me fix this ? 

Arduino: 1.8.1 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

O:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware O:\Program Files (x86)\Arduino\hardware -hardware W:\Users\Steppo\AppData\Local\Arduino15\packages -hardware W:\Users\Steppo\Documents\Arduino\hardware -tools O:\Program Files (x86)\Arduino\tools-builder -tools O:\Program Files (x86)\Arduino\hardware\tools\avr -tools W:\Users\Steppo\AppData\Local\Arduino15\packages -built-in-libraries O:\Program Files (x86)\Arduino\libraries -libraries W:\Users\Steppo\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10801 -build-path W:\Users\Steppo\AppData\Local\Temp\arduino_build_540447 -warnings=none -prefs=build.warn_data_percentage=75 -verbose W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\libraries\LiOr_WiFi_Projekt\Empfaenger\ESP32_WS2812B_WiFi_17_12_12\ESP32_WS2812B_WiFi_17_12_12.ino
O:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware O:\Program Files (x86)\Arduino\hardware -hardware W:\Users\Steppo\AppData\Local\Arduino15\packages -hardware W:\Users\Steppo\Documents\Arduino\hardware -tools O:\Program Files (x86)\Arduino\tools-builder -tools O:\Program Files (x86)\Arduino\hardware\tools\avr -tools W:\Users\Steppo\AppData\Local\Arduino15\packages -built-in-libraries O:\Program Files (x86)\Arduino\libraries -libraries W:\Users\Steppo\Documents\Arduino\libraries -fqbn=espressif:esp32:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10801 -build-path W:\Users\Steppo\AppData\Local\Temp\arduino_build_540447 -warnings=none -prefs=build.warn_data_percentage=75 -verbose W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\libraries\LiOr_WiFi_Projekt\Empfaenger\ESP32_WS2812B_WiFi_17_12_12\ESP32_WS2812B_WiFi_17_12_12.ino
Using board 'esp32' from platform in folder: W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10801 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\variants\esp32" "W:\Users\Steppo\AppData\Local\Temp\arduino_build_540447\sketch\ESP32_WS2812B_WiFi_17_12_12.ino.cpp" -o "nul"
"W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IW:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IW:\Users\Steppo\Documents\Arduino\ha [... cut ...]
   ^

Bibliothek Adafruit_NeoPixel-master in Version 1.1.2 im Ordner: W:\Users\Steppo\Documents\Arduino\libraries\Adafruit_NeoPixel-master  wird verwendet
Bibliothek Preferences in Version 1.0 im Ordner: W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\libraries\Preferences  wird verwendet
Bibliothek WiFi in Version 1.0 im Ordner: W:\Users\Steppo\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

Buffalchill, 2017-12-15:
I got 3 PCs... 2 of them were able to compile it. Then i had to reinstall the whole ESP32 on one PC.

now only 1 PC is able to compile it. 

I think its the ESP IDF version that makes it uncompilable... 

1. how can i see which ESP32 IDF release ist currently installed ?
2. how can i downgrade the ESP32 IDF to an older release ?

Buffalchill, 2017-12-15:
Is it possible to install an older version of the Arduino Core ?
How can i do it ?

Buffalchill, 2017-12-16:
they helped me in the gitter chat nvm

closed 2017-12-16 by Buffalchill

williamesp2015, 2020-04-26:
i have problem with new version :
*** My problem was brownout detect when starting to enable WiFi.mode(WIFI_MODE_AP);
I added
 #include "soc/soc.h" 
#include "soc/rtc_cntl_reg.h"

void setup()
{
....
uint32_t brown_reg_temp = READ_PERI_REG(RTC_CNTL_BROWN_OUT_REG); //save WatchDog register
WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector
 Serial.println(F("WIFI MODE AP"));
 WiFi.mode(WIFI_MODE_AP);
 WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, brown_reg_temp); //enable brownout detector

The ERROR IS:
.....platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32/esp_assert.h:31:44: error: 'failed_compile_time_assert'  was not declared in this scope
                 failed_compile_time_assert();