espressif/arduino-esp32 #928: EEPROM: EEPROM.end() crashes and reboots

GrooverFromHolland, 2017-12-15:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation/update date:			17/November/2017
IDE name:						VisualStudio 201 with visualMicro
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
First i have to mention that i cannot use the EspExceptionDecoder because my sketch  is not compiling in the Arduino IDE: error: 'class BLEClient' has no member named 'isConnected'. **This is not related to my problem**.
My problem is that if i want to write to EEPROM, after EEPROM.commit() or EEPROM.end() the application crashes and reboots.
All EEPROM examples I could find work OK.
### Sketch:
```
/**
* A BLE client with IR receiver
*/
#include <dummy.h>
#include <IRremote.h>
#include <boarddefs.h>
#include "BLEDevice.h"
#include"bt.h"
#include "BLEScan.h"
#include "EEPROM.h"

//IR
int RECV_PIN = 22;
IRrecv irrecv(RECV_PIN);
decode_results results;
//BLE
static BLEUUID serviceUUID("0000ffe0-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in:
static BLEUUID     charUUID("0000ffe1-0000-1000-8000-00805f9b34fb");
static BLEClient*  pClient;
static BLEAddress  *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
const byte stx = 2;
const byte etx = 3;
const byte dle = 10;
byte brite = 0;
int hue = 0;
int step = 2;
byte sendbufferHue[7] = { stx,30,dle,brite,0,0,etx };;
byte sendbufferBrite[7] = { stx,1,dle,brite,0,0,etx };;
boolean powerSet = false;
volatile bool clearToSend = true;
volatile bool tosend = true;
int briteUpDown = 1;
int hueUpDown = 1;
int countHue = 0;
bool valuesChanged = false;
//timer and eeprom
ulong startMillis = 0;
bool timerIsRunning = false;
bool safeToWrite = false;
byte packetBrite[7];
byte packetHue[7];

//packet = stx;cmd;dle;byte;byte;byte;etx (fixt length = 7)

class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
	/**
	* Called for each advertising BLE server.
	*/
	void onResult(BLEAdvertisedDevice advertisedDevice)
	{
		Serial.print("BLE Advertised Device found: ");
		Serial.println(advertisedDevice.toString().c_str());
[... 328 lines cut ...]
### Debug Messages:
Opening port
Port open
218148001
bright is: 5
to send = 1
[D][BLERemoteCharacteristic.cpp:543] writeValue(): >> writeValue(), length: 7
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_WRITE_CHAR_EVT
[D][BLERemoteCharacteristic.cpp:565] writeValue(): << writeValue
 here
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400dafbc  PS      : 0x00060034  A0      : 0x400851e8  A1      : 0x3ffc0c80  
A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x40088e98  
A6      : 0x00000000  A7      : 0x3ffd50e0  A8      : 0x800811f0  A9      : 0x3ff5f024  
A10     : 0x3ffc10dc  A11     : 0x20000000  A12     : 0x00000400  A13     : 0x3ffda2a0  
A14     : 0x00000000  A15     : 0x00290000  SAR     : 0x00000015  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffff8  

Backtrace: 0x400dafbc:0x3ffc0c80 0x400851e5:0x3ffc0ca0 0x4008650e:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

```

stickbreaker, 2017-12-15:
@GrooverFromHolland you might want to do a little reading about how this device functions,  specifically review ATTR_IRAM 

You are registering a callback that is not guaranteed to be loaded while using `EEPROM()` which has to disable instruction caching while it operates. 

Chuck.

GrooverFromHolland, 2017-12-15:
@stickbreaker. I googled ATTR_IRAM and couldnot find any information on that, also checked all used libraries and did not find ATTR_IRAM! So how is this related to my problem?

stickbreaker, 2017-12-15:
@GrooverFromHolland do a Google search on [readthedocs](https://esp-idf.readthedocs.io/)

The ESP32 only has 520kB of RAM, a small part ~128kB is the instruction cache. It loads program code from the 4MB FLASH, then executes from this RAM.  The `EEPROM()` library emulates the EEPROM in a AVR processor by allocating a RAM buffer of the Requested Size (multiple of 4096 bytes, because the FLASH has to ERASE 4kB at a time). It then applies your `EEPROM.write()` to this RAM buffer, when you `commit()` it Disables Cache because it has to Take control of the FLASH, (background code loading is frozen) Issues the Flash Erase command (which can take a long Time, tens to hundreds of ms). Since this task is held waiting for the Hardware, FreeRTOS finds another task that it can release, or in your case, BLE issues a callback.  But, since you did not tell the compiler to always keep the callback in RAM, the Callback code is still in the FLASH which is eraseing a sector for `EEPROM()`. Since the background instruction loading function needs to use the cache and the FLASH to service this callback.  It explodes.  

Callbacks, Interrupts, any code that can execute out of order needs to always reside in the limited 128kB instructionRam.  So any code that can be called out of order needs ATTR_IRAM which forces the compiler to keep it in RAM forever.  AND, every call from that code must ALSO be in RAM at all times.  So, be parsimonious about which code you call. 

```c++
class ATTR_IRAM MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
```
Chuck.

GrooverFromHolland, 2017-12-15:
@stickbreaker. I tried to implement you suggestion, but i don't understand much c++. When i build it I am getting a build error: Ble_remote.ino: 52:45: error: expected initializer before ':' token
 class ATTR_IRAM MyAdvertisedDeviceCallbacks *: public BLEAdvertisedDeviceCallbacks.
How to resolve this?

stickbreaker, 2017-12-15:
@GrooverFromHolland, Sorry working on Window XP Pascal program right now,  Brain Fade,
Try IRAM_ATTR

```c++
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
	/**
	* Called for each advertising BLE server.
	*/
	void IRAM_ATTR onResult(BLEAdvertisedDevice advertisedDevice)
	{
		Serial.print("BLE Advertised Device found: ");
		Serial.println(advertisedDevice.toString().c_str());
		// We have found a device, let us now see if it contains the service we are looking for.
		if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID))
		{
			Serial.print("Found our device!  address: ");
			int rssi = advertisedDevice.getRSSI();
			Serial.print("Rssi: ");
			Serial.println(rssi);
			advertisedDevice.getScan()->stop();
			pServerAddress = new BLEAddress(advertisedDevice.getAddress());
			doConnect = true;
		}
	}
};
```
Chuck.

GrooverFromHolland, 2017-12-15:
@stickbreaker. No luck wth that:
Compiling 'Ble_remote' for 'DOIT ESP32 DEVKIT V1'
 
Ble_remote.cpp.o*: In function MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)
Ble_remote.ino:59: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x3)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x4)
Ble_remote.ino:59: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x6)  dangerous relocation  l32r  literal placed after use  .literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice
Ble_remote.ino:62: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x44)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x8)
Ble_remote.ino:64: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xdf)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0xc)
Ble_remote.ino:66: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xf2)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x10)
Ble_remote.ino:69: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x139)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x14)
Ble_remote.ino:70: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x150)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x18)
Ble_remote.ino:59: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xc)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x1c)
Ble_remote.ino:60: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x16)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x20)
Ble_remote.ino:60: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x20)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x24)
 
Ble_remote.cpp.o*: In function std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string()

Error linking for board DOIT ESP32 DEVKIT V1
basic_string.h:544: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x28)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x28)
Build failed for project 'Ble_remote'
 
Ble_remote.cpp.o*: In function MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice)
Ble_remote.ino:62: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x30)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x2c)
Ble_remote.ino:62: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x3e)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x30)
Ble_remote.ino:62: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xd6)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x34)
Ble_remote.ino:64: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xe4)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x38)
Ble_remote.ino:65: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xec)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x3c)
Ble_remote.ino:66: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0xfa)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x40)
Ble_remote.ino:67: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x106)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x44)
Ble_remote.ino:68: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x10e)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x48)
Ble_remote.ino:68: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x114)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x4c)
Ble_remote.ino:69: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x11c)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x50)
Ble_remote.ino:69: (.iram1[MyAdvertisedDeviceCallbacks  onResult(BLEAdvertisedDevice)]+0x127)  dangerous relocation  l32r  literal placed after use  (.literal._ZN27MyAdvertisedDeviceCallbacks8onResultE19BLEAdvertisedDevice+0x54)
 
collect2.exe*: error: ld returned 1 exit status

stickbreaker, 2017-12-15:
@GrooverFromHolland , looks like you are going to have to attach it from the other direction.
separate the `EEPROM()` functions away from BLE

I just spent some time on your code, can you explain what this is suppose to do?
```c++
if (pClient->isConnected() == true) {
  digitalWrite(LED_BUILTIN, HIGH);
  }
else{
  digitalWrite(LED_BUILTIN, LOW);
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->start(40);
  }
```

It's at the end of your main loop.

This local variable `pBLEScan()` only exists for a very very short time?

Chuck.

GrooverFromHolland, 2017-12-15:
@stickbreaker. That code is to show me that there is a connection with the device i am targetting and if the conection is lost for some reason: rescan  and try to reconnect.

stickbreaker, 2017-12-16:
@GrooverFromHolland my question was that you are creating a pointer to an object, then discarding the pointer.  The pointer goes out of scope,  does the dynamic object get destroyed? created by `BLEDevice::getScan()` 

Dumb question.  I just read through BLEDevice.cpp. (ignore it)

If you don't do the `EEPROM()` calls does everything else work?

Chuck.

GrooverFromHolland, 2017-12-16:
@stickbreaker. Yes, It worksperfect without Eeprom.

GrooverFromHolland, 2017-12-17:
**Update:** Tested with Preferences.h instead of EEPROM.h and got the same problem. 
This means it is not likely that Preferences.h nor EEPROM.h is the cause.
After removing IRremote.h and simulating the IRremote code everyting worked fine again, even with both Preferences and EEPROM.
Now I must find another IRremote library that is compatible with EEPROM.h and BLE.
Any help on this will be appreciated.

beegee-tokyo, 2017-12-17:
Didn't try on ESP32 yet, but I was using [IRLib2](https://github.com/cyborg5/IRLib2) on an ESP8266 project and it works fine there.

mitchsf, 2017-12-17:
I gave up on IRRemote, and I also tried an RMT library that did not work reliably. I ended up using an ATTiny85 running IRRemote, and a serial connection to the ESP32. Obviously it would have been ideal to have a working IR library running on the ESP32.

GrooverFromHolland, 2017-12-17:
**UPdate:** Finally after a lot if digging in IRremote I found a solution that is working. 
I deleted everything in _IrSend.ccp_ (maybe not necessary, but  I do not need it ),
 modified _IRremote_h_ so ` void  enableIRIn ();` can have a bool as parameter: 
`void  enableIRIn (bool enable );`, 
modified _irRecv.ccp_ to:

```
`void  IRrecv::enableIRIn(bool enable)
{
	// Interrupt Service Routine - Fires every 50uS
#ifdef ESP32
	// ESP32 has a proper API to setup timers, no weird chip macros needed
	// simply call the readable API versions :)
	// 3 timers, choose #1, 80 divider nanosecond precision, 1 to count up

	if (enable) {

		timer = timerBegin(1, 80, 1);
		timerAttachInterrupt(timer, &IRTimer, 1);
		// every 50ns, autoreload = true
		timerAlarmWrite(timer, 50, true);
		timerAlarmEnable(timer);
}
	else {

		timerEnd(timer);
		timerDetachInterrupt(timer);
	}`
```
Before writing to eeprom or preferences I use `irrecv.enableIRIn(false);` and after `irrecv.enableIRIn(true);`
Also `irrecv.enableIRIn(true);` must be called after there is a BLE connection.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

skrodahl, 2018-12-07:
GrooverFromHolland's solution works, but his instructions are not correct. I've written the solution here, together with another problem that prevents IRremote.h from compiling on ESP32:
https://hackaday.io/project/46280-muffsy-stereo-relay-input-selector/log/156747-full-functionality-rotational-encoder-and-remote-control