espressif/arduino-esp32 #929: Can I use xTaskCreatePinnedToCore() with Core 0?

huming2207, 2017-12-17:
Hi all,

I've got a question about ```xTaskCreatePinnedToCore()``` function and  I would like to ask that if I create a task **on the core where Arduino framework doesn't run on**. Right now in Arduino framework that core should be Core 0.

I googled it around and I've found this issue: https://github.com/espressif/arduino-esp32/issues/436. It seems like that all serial related methods must be run on the "arduino running core", which is Core 1 currently.

So guys, if I put some tasks such as ``ESPWebServer::handleClient()`` to Core 0 with no hardware-related methods like `Serial` or `Wire`, will it works perfectly?

Regards,
Jackson

beegee-tokyo, 2017-12-17:
Just my configuration using xTaskCreatePinnedToCore()
I2C device (Wire) on core 0. Debug output over Serial and result posting over an MQTT client.
MQTT client (HTTP Client) on core 0.. Debug output over Serial.
HTTP client (HTTP GET) on core 0. Debug output over Serial and result posting over an MQTT client.
Exception:
Bitbanging DHT11 function on core 1. When I try to switch this to core 0 it fails with timeouts and checksum errors.

I tested this configuration by trial and error ;)

lonerzzz, 2017-12-19:
@huming2207 You can create a task pinned to either core but unless you expressly know the reason for pinning, you are better off letting the task float so that Arduino and the WiFi portions can best do what they need to. I have both in my applications.

huming2207, 2017-12-19:
Hi guys, thanks for your reply. I'll try it again on my implementations.

closed 2017-12-19 by huming2207

SalmaG98, 2019-08-22:
@beegee-tokyo I sent you an email on bernd@giesecke.tk to explain the same issue I am having trying to work on a project similar to yours