espressif/arduino-esp32 #931: Abnormal disconnection of HTTPClient

mytrain, 2017-12-17:
### Hardware:
Board: ESP32 Dev Module
Core Installation/update date: 29/Nov/2017
IDE name: Arduino IDE
SDK Version: v3.0-dev-1325-ge5b2c1cf

### Description:
When opening 2 HTTPClient to 2 different URL, completing the reading of a seconde one witch connection status of both client to 'disconnected'

### Sketch:
[issue931.txt](https://github.com/espressif/arduino-esp32/files/1565983/issue931.txt)

### Outputs:
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 2 - STA_START
6
6
6
6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 4 - STA_CONNECTED
6
6
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 7 - STA_GOT_IP
[V][HTTPClient.cpp:140] beginInternal(): url: http://www.google.com
[D][HTTPClient.cpp:181] beginInternal(): host: www.google.com port: 80 url: 
[D][HTTPClient.cpp:850] connect():  connected to www.google.com:80
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'HTTP/1.0 404 Not Found'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Type: text/html; charset=UTF-8'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Referrer-Policy: no-referrer'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Length: 1564'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Date: Sun, 17 Dec 2017 19:36:03 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:968] handleHeaderResponse(): code: 404
[D][HTTPClient.cpp:971] handleHeaderResponse(): size: 1564
[V][HTTPClient.cpp:140] beginInternal(): url: http://www.youtube.com
[D][HTTPClient.cpp:181] beginInternal(): host: www.youtube.com port: 80 url: 
[D][HTTPClient.cpp:850] connect():  connected to www.youtube.com:80
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'HTTP/1.0 404 Not Found'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Type: text/html; charset=UTF-8'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Referrer-Policy: no-referrer'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Content-Length: 1564'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: 'Date: Sun, 17 Dec 2017 19:36:03 GMT'
[V][HTTPClient.cpp:938] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:968] handleHeaderResponse(): code: 404
[D][HTTPClient.cpp:971] handleHeaderResponse(): size: 1564
>>>[1] httpFile1[connected] httpFile2[connected] lenFile[1564]]
>>>[2] httpFile1[connected] httpFile2[connected] lenFile[540]]
>>>[3] httpFile1[connected] httpFile2[connected] lenFile[540]]
>>>[2] httpFile1[connected] httpFile2[disconnected] lenFile[0]]
>>>[3] httpFile1[disconnected] httpFile2[disconnected] lenFile[0]]

On last line I do not understand why the httpFile1 instance goes to disconnected.

lonerzzz, 2017-12-19:
@mytrain Can you provide a code sample to let us better help you?

mytrain, 2017-12-19:
Arduino sketch is in attachement issue931.txt (sadly it is not possible to upload an ino extension file on github 👎 )

closed 2018-04-19 by everslick