espressif/arduino-esp32 #932: WiFi.macAddress()

tackelua, 2017-12-18:
### Hardware:
Board:							Doit ESP32 DevKit V1
Core Installation/update date:		6 Dec 2017
IDE name:						Microsoft Visual Studio + Visual Micro
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
Describe your problem here

### Sketch:
```cpp

#include <Arduino.h>
#include <WiFi.h>
void setup()
{
	Serial.begin(115200);
	String mac = WiFi.macAddress();
	Serial.println(mac);
	delay(100);
	mac = WiFi.macAddress();
	Serial.println(mac);
	delay(50);
	Serial.println(WiFi.macAddress());
}

void loop()
{
	delay(0);
}

```

### Debug Messages:
```
Opening port
Port open
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
E (20) wifi: esp_wifi_get_mac 1215 wifi is not init
80:AA:FC:3F:44:23
E (121) wifi: esp_wifi_get_mac 1215 wifi is not init
AA:00:00:00:FC:00
E (171) wifi: esp_wifi_get_mac 1215 wifi is not init
00:00:00:00:00:00

```

beegee-tokyo, 2017-12-18:
If you want the MAC address before WiFi.mode(xxx) and WiFi.begin() is called you need to use something like:
```cpp
#include "esp_system.h"
String getMacAddress() {
	uint8_t baseMac[6];
	// Get MAC address for WiFi station
	esp_read_mac(baseMac, ESP_MAC_WIFI_STA);
	char baseMacChr[18] = {0};
	sprintf(baseMacChr, "%02X:%02X:%02X:%02X:%02X:%02X", baseMac[0], baseMac[1], baseMac[2], baseMac[3], baseMac[4], baseMac[5]);
	return String(baseMacChr);
}
```

tackelua, 2017-12-18:
Thank you @beegee-tokyo

closed 2017-12-19 by lonerzzz