espressif/arduino-esp32 #933: goouuu_esp32 i didnt find this version of esp 32 in the board list please insert it

omarxa, 2017-12-18:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

copercini, 2017-12-18:
I have this board, it works as "ESP32 Dev Module"

closed 2017-12-18 by copercini

omarxa, 2017-12-18:
but its not defining the bult in led