espressif/arduino-esp32 #935: Time.h and Unix timestamp

bill-orange, 2017-12-19:
### Hardware:
Board:							ESP32 TTGO Module
Core Installation/update date:	        unknown
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
I notice that a few days ago time.h was added to the library.  Good news.  The example sketch works fine.  I notice, however, that in the time.h structure there is tm_hour, tm_min, etc, but no unix time.  Is there an easy way to get the unix timestamp with existing libraries?

MarkyAD, 2017-12-20:
Setup the same as example sketch then:

```javascript
time_t now;

time(&now);
Serial.println(now);

bill-orange, 2017-12-20:
Works like a charm. Thanks.  I am still a bit confused by both your suggestion and the example sketch.  I don't fully understand how time.h works which leaves me confused as to what portions of the code should go in setup(), be declared before setup()  and a called within loop().  Right now I have everything pretty much in the function called by loop(), like the example, but it 'looks' wrong.  Is this correct?

const char* ntpServer = "pool.ntp.org";

unsigned long getTime() { 
  time_t now;
  struct tm timeinfo;
  if (!getLocalTime(&timeinfo)) {
    Serial.println("Failed to obtain time");
    return(0);
  }
  configTime(0, 0, ntpServer);
  time(&now);
  return now;
}

MarkyAD, 2017-12-20:
getLocalTime() calls time() to fetch the unix timestamp and localtime_r() to convert that to local time, filling in the struct tm elements (tm_hour, tm_min, etc). It also has a timeout that waits for the time to be initialized, call it once in setup() to wait for the clock to be set. After that, if all you need is the unix timestamp you only need to call time().

bill-orange, 2017-12-22:
Got it! Thanks.

closed 2017-12-22 by bill-orange