espressif/arduino-esp32 #936: load cell with HX711

emsoft1, 2017-12-20:
my issues is i have cod for load cell but when  i loaded to the EPS32 program is working but ESP32 is not get date from HX711  but when i connect HX711 to arduino nano works perfectly  i do not know whats happens in this devises ? what should i do ? any one work with this  2 board  ?

beegee-tokyo, 2017-12-21:
Maybe this is your problem [ESP32 issue, clock speed too fast #75](https://github.com/bogde/HX711/issues/75).
HX711 communication is kind of a bit-banging and libraries I found seem not take into account the CPU speed. An Arduino Nano runs with 8 or 16MHz, an ESP32 runs with 240MHz, therefore the produced clock might be much too fast for the HX711. I saw [lemio/HX711](https://github.com/lemio/HX711) which seems to try to adapt the clock speed. Maybe you can give it a try.

And a tip. For future issues please give more details, code example, used IDE, used libraries, ... to make it easier for others to help.

rokrodic, 2018-03-16:
Yes. I have been working with ESP32 and HX711 and the main problem is the shiftIn function. It is too fast for HX711. See the HX specification of the protocol. One way is to add some delays into shiftIn, but this is not advisable if you intend to upgrade your ESP32 installation someday :-) The other way is to slow down ESP32 - include something like:
#ifdef ARDUINO_ESP32_DEV
  #include <soc/rtc.h>
  rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
#endif
Check: https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/rtc_clk.c

rokrodic, 2018-03-16:
But the final solution is that Espressif enables option for us to set the delays in the ShiftIn function itself!!! This is a feature request.

squix78, 2018-04-22:
I'm also fighting with this issue. Reducing CPU frequency resolves the problem but I need to put the device to sleep and it hangs, if I set the CPU frequency again down to 80MHz after sleep. I'm not 100% if it really is the shiftIn function. I tried lemio's code and I still have these issues. I attached an logic analyzer and you can find the screenshots of it here: https://github.com/bogde/HX711/issues/75 There seems to be a strange glitch every now and then. Anyone of you guys has an idea what this could be?

amotl, 2019-02-17:
Hi there,

we have been able to give the [HX711 library](https://github.com/bogde/HX711) some love based on all the valuable contributions and suggestions there. https://github.com/bogde/HX711/pull/123 resp. [1] might already solve your problem, we will appreciate any feedback about this.

With kind regards,
Andreas.

[1] https://github.com/hiveeyes/HX711/tree/spring-cleaning

amotl, 2019-08-01:
Hi Arduino/ESP32 community,

following up on https://github.com/bogde/HX711/pull/123 referenced above which folds many valuable contributions from the community into the library, we have been able to confirm hardware support for the following devices/architectures with the improved HX711 library by @bogde.

- ATmega328: Arduino Uno
- ESP8266: WeMos D1 mini, Adafruit HUZZAH
- ESP32: ESP32 DEVKIT V1, Heltec WiFi Kit 32, Adafruit Feather HUZZAH32
- STM32 F1 (Cortex-M3): STM32F103C8T6 STM32 Blue Pill Board

The library is also available from the Arduino Library Manager these days.

With kind regards,
Andreas.

-- https://github.com/bogde/HX711#hardware-support

closed 2019-10-14 by stale[bot]