espressif/arduino-esp32 #938: problems when uploading dwm1000 examples to esp 32

Ruben-26, 2017-12-21:
Greetings, thank you very much for helping me solve this problem:
I have a plate esp32 of adafruit.
I managed to load a program to turn on and off a led accessing a server. Now I try to load the examples of dwm 1000, but in the plate esp 32 does not upload them, and in an arduino one yes.
the error that appears to me when copying on a tarjeya esp32 is the following:

:\Users\Usuario\Documents\Arduino\libraries\arduino-dw1000-master\src\DW1000.cpp: In static member function 'static void DW1000Class::begin(uint8_t, uint8_t)':

C:\Users\Usuario\Documents\Arduino\libraries\arduino-dw1000-master\src\DW1000.cpp:172:6: error: 'class SPIClass' has no member named 'usingInterrupt'

  SPI.usingInterrupt(digitalPinToInterrupt(irq)); // not every board support this, e.g. ESP8266

      ^

exit status 1
Error compilando para la tarjeta ESP32 Dev Module.
Un subdirectorio de su 'cuaderno de bocetos' no es una librería válida
Librería inválida encontrada en C:\Users\Usuario\Desktop\Arduino\libraries\arduino-esp32-master: C:\Users\Usuario\Desktop\Arduino\libraries\arduino-esp32-master
Librería inválida encontrada en C:\Users\Usuario\Documents\Arduino\libraries\arduino-esp32-master: C:\Users\Usuario\Documents\Arduino\libraries\arduino-esp32-master
Librería inválida encontrada en C:\Users\Usuario\Documents\Arduino\libraries\msys32:

everslick, 2018-04-19:
Looks like this library is just not compatible with ESP32. you could try reaching out to the library maintainer and ask for support.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

ganeshx-o, 2019-01-02:
> Greetings, thank you very much for helping me solve this problem:
> I have a plate esp32 of adafruit.
> I managed to load a program to turn on and off a led accessing a server. Now I try to load the [...]

just go to DW1000.cpp ,inside  void DW1000Class::begin(uint8_t irq, uint8_t rst)  just replace #ifndef ESP8266 as #ifndef ESP32.....now the code will compile succesfully