espressif/arduino-esp32 #940: BLE server - only one client can connect?

beegee-tokyo, 2017-12-22:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		22/dec/2017
IDE name:						Platform.io Arduino framework
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Using @nkolban BLE server example, I created successful a BLE server that I can connect to from my Android tablet as a client (with [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)) and receive data from the ESP32.
But if I try connect a second client, this second client cannot connect.
The error message on the client is **_Error 133 (0x85) GATT ERROR_**.
I set **_-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG_** before building the ESP32 code. In the log I can see the first client connecting and receiving data and notifications. But the log doesn't show anything about a second client trying to connect.

From what I read about BLE, it should be possible to connect several clients to one server. Or did I misunderstand something or miss something in my code?
Sketch below includes the callbacks for server and characteristics. BLE is initialized at the end of **`setup()`** by a call to **`initBLE()`**.

### Sketch:
```cpp
#include "setup.h"
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

// List of Service and Characteristic UUIDs
#define SERVICE_UUID          0x181A
#define NOTIFICATION_UUID     0x2A08
#define TEMP_UUID             0x2A1F
#define HUMID_UUID            0x2A6F
#define STATUS_UUID           0x2A3D

/** Characteristic for client notification */
BLECharacteristic *pCharacteristicNotify;
/** Characteristic for temperature in celsius */
BLECharacteristic *pCharacteristicTemp;
/** Characteristic for humidity in percent */
BLECharacteristic *pCharacteristicHumid;
/** Characteristic for environment status */
BLECharacteristic *pCharacteristicStatus;

/** BLE Advertiser */
BLEAdvertising* pAdvertising;
/** BLE Service */
BLEService *pService;
/** BLE Server */
BLEServer *pServer;

/** Flag if a client is connected */
bool bleConnected = false;
/** Last temperature reading */
float bleTemperature;
/** Last humidity reading */
float bleHumidity;
/** Last environment status */
String bleStatus;

/**
 * MyServerCallbacks
 * Callbacks for client connection and disconnection
 */
class MyServerCallbacks: public BLEServerCallbacks {
  // TODO this doesn't take into account several clients being connected
    void onConnect(BLEServer* pServer) {
      bleConnected = true;
      addMqttMsg("debug", "[INFO] " + digitalTimeDisplaySec() + " BLE connected", false);
    };

    void onDisconnect(BLEServer* pServer) {
      bleConnected = false;
      addMqttMsg("debug", "[INFO] " + digitalTimeDisplaySec() + " BLE disconnected", false);
    }
[... 166 lines cut ...]
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1748] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x31]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_READ_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_READ_EVT
[D][BLEUtils.cpp:1741] dumpGattServerEvent(): [conn_id: 0, trans_id: 18, bda: f8:00:3d:62:49:be, handle: 0x31, is_long: 1, need_rsp:1]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_READ_EVT
[D][BLECharacteristic.cpp:324] handleGATTServerEvent(): - Testing: 0x31 == 0x2a
[D][BLECharacteristic.cpp:324] handleGATTServerEvent(): - Testing: 0x31 == 0x2d
[D][BLECharacteristic.cpp:324] handleGATTServerEvent(): - Testing: 0x31 == 0x2f
[D][BLECharacteristic.cpp:324] handleGATTServerEvent(): - Testing: 0x31 == 0x31
[D][BLECharacteristic.cpp:629] setValue(): >> setValue: length=50, data=436f6d666f72743a20546f6f20486f742050657263657074696f6e3a20536f6d65686f7720756e636f6d666f727461626c65, characteristic UUID=00002a3d-0000-1000-8000-00805f9b34fb
[D][BLECharacteristic.cpp:636] setValue(): << setValue
[D][BLECharacteristic.cpp:352] handleGATTServerEvent(): Sending a response (esp_ble_gatts_send_response)
[D][BLECharacteristic.cpp:387] handleGATTServerEvent():  - Data: length=6, data=727461626c65, offset=44
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:80] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 3] ... ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1611] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEUtils.cpp:1748] dumpGattServerEvent(): [status: ESP_GATT_OK, handle: 0x31]
[D][BLEServer.cpp:175] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_RESPONSE_EVT
[D][BLEServer.cpp:287] handleGATTServerEvent(): << handleGATTServerEvent
```
**No output on log when second client tries to connect.**

beegee-tokyo, 2017-12-23:
Discussion continued on [nkolban/esp32-snippets](https://github.com/nkolban/esp32-snippets/issues/307) with @nkolban.
Quick and dirty patch created as a [gist](https://gist.github.com/beegee-tokyo/846d71fca439b81e8aa848a2d4fac989)

closed 2017-12-23 by beegee-tokyo

RottenVanSuti, 2018-05-18:
@beegee-tokyo Hi, I have replaced the patch you provide, but errors appears for BLEDescriptor.h, BLEDescriptor.cpp and BLE2904.h, BLE2904.cpp when trying to compile. The image:
[image]

I'm not so expert yet, I'm not sure why those errors appears. If you could help me, I would aprecciate it, please.

beegee-tokyo, 2018-05-19:
@RottenVanSuti since I wrote the patch the BLE library might have gotten some updates. I will compare the patch with the current version of the BLE lib. But I will not be able to this before Monday.

beegee-tokyo, 2018-05-19:
@RottenVanSuti if you dont use notifications, just restarting advertising is sufficient. The patch is only necessary if clients subscribe to server notifications. Without the patch only the first registered client receives the notification.
When several clients send data to the same characteristic, the last write is saved. Clients writing before will not know that their value was overwritten. The server (as far as I understand) cannot 'send' data to the clients, he can send notifications to subscribed clients and they then read the characteristics new value.

RottenVanSuti, 2018-05-19:
Yes, you are right, I mean notifications. Ok, so with the patch, sending notifications to clients subscribed, is solved?

beegee-tokyo, 2018-05-19:
Yes it worked for me. Here is my [application](https://desire.giesecke.tk/index.php/2018/02/02/esp32-ble-server/)

beegee-tokyo, 2018-05-19:
What you can do without notifications is to read characteristics frequently to check if the value has changed.

RottenVanSuti, 2018-05-19:
Ok, thank you, I've already seen your application, it is great for guidance. I'll keep trying to solve the errors that appears when patching the ble library, so I can notify the clients. I'll wait for you, if you can check if something has been modified to the ble library, although I didn't find anything different. Thanks for your time

beegee-tokyo, 2018-05-21:
@RottenVanSuti had a quick look and compared the files from my gist and the current files (ESP32 staging version) and found a lot of differences. Also looking on your error messages, you are using descriptor BLE2904. 
It is a long time since I played around with this, but I think that you need to use BLE2902 descriptor for notifications and indications. Looking into BLECharacteristic.cpp I can see that it only supports notifications and indications on descriptor BLE2902.

Basically, what I did was to expose the connection ID `conn_id` to the descriptors getValue and setValue functions, to support multiple clients.

I personally gave up on the multiple client support, as it is officially not supported by BLE.

RottenVanSuti, 2018-05-25:
Ok, thank you @beegee-tokyo  . So, That what you did about to expose the connection ID conn_id,  did not work as you expected ?. I've tried using the server example and client example, where two ESP32 are clients and one is server and the clients connect to the ESP32 as server. However, there is an issue because sometimes the client or server (I'm not sure which one) hangs, or throws the following error: BT: l2cble_process_conn_update_evt: Error status.

RottenVanSuti, 2018-05-25:
yes, I decided to use BLE2902 and many errors disappeared. Btw sorry for my late response

beegee-tokyo, 2018-05-26:
@RottenVanSuti it worked as expected, but I gave up on it because it seems its not supported by the BLE standards.

sagarvegg, 2019-06-10:
I have face problem with BLE server multi connect
only able to connect up to four device to single esp32 wrover
I change sdkconfig "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN" = 9 as well "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN__EFF" = 9 for up to nine device connect to ble server
and also changed "CONFIG_BT_ACL_CONNECTIONS" = 7 for seven server connected to single client

but don't connect more than 4 device to single ble server

after resetting above sdkconfig parameters only connect 3 device to esp32 wrover

if any one have solution???

chegewara, 2019-06-10:
@sagarvegg i thought you are using esp-idf, but now im not sure. Could you clarify this?

sagarvegg, 2019-06-10:
@chegewara yes I uses esp-idf for arduino and change my sdkconfig file for multi connect..
I change sdkconfig.h file parameters at location "..\Arduino\hardware\espressif\esp32\tools\sdk\include\config".

I use BLE [client + server] at same instance; before changes in sdkconfig,h file I only able to connect 3 multi server connection to 3 different device which has common GATT server Configure by nRF connect; but after changes sdkconfig.h only able to connect 4 device to single ESP32 wrover ... but configuration says that BLE maximum 9 server and 7 client connection possible for esp32 i.e.: "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN" = 9 & "CONFIG_BT_ACL_CONNECTIONS" = 7

atanisoft, 2019-06-10:
@sagarvegg changing the sdkconfig for arduino-esp32 in the location you mention won't help as the IDF buts were already compiled with the existing settings. To modify these you would need to recompile the IDF side and at which point it would be better to use Arduino as an IDF component.

chegewara, 2019-06-10:
Then last one question, how do you change sdkconfig? Because editing it wont work, it will be overwritten, you have to use `make menuconfig`.

sagarvegg, 2019-07-23:
I solved problem with BLE server multi connect
Now able to connect more than four device to single esp32 wrover
I change sdkconfig "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN" = 9 as well "CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN__EFF" = 9 for up to nine device connect to ble server
and also changed "CONFIG_BT_ACL_CONNECTIONS" = 7 for seven server connected to single client

and generate sdkconfig file again via "Using as ESP-IDF component" for arduino...

AS123-y, 2021-10-02:
@sagarvegg i want to implement a network using ESP32 , i want to be able to connect one ESP32 to many ESP32s 
i already use the client/server example , but i was able to connect only two devices .

any help or guidance please