espressif/arduino-esp32 #944: OTA requires an ability to specify the interface when used with ETH rather than WIFI

dirkx, 2017-12-24:
When using OTA in combination with Ethernet (rather than Wifi) - one need sto explictly sets
the interface of mDNS (as the latter does not auto-attach to the active interface).

So below is required to make OTA work on Ethernet (in conjuction with 	https://github.com/espressif/esp-idf/issues/1431 if auto discovery is to be made to work too).

Dw.

```
diff --git a/libraries/ArduinoOTA/src/ArduinoOTA.cpp b/libraries/ArduinoOTA/src/ArduinoOTA.cpp
index 2516420..af46091 100644
--- a/libraries/ArduinoOTA/src/ArduinoOTA.cpp
+++ b/libraries/ArduinoOTA/src/ArduinoOTA.cpp
@@ -97,7 +97,7 @@ ArduinoOTAClass& ArduinoOTAClass::setMdnsEnabled(bool enabled){
     return *this;
 }
 
-void ArduinoOTAClass::begin() {
+void ArduinoOTAClass::begin(tcpip_adapter_if_t tcpip_if) {
     if (_initialized){
         log_w("already initialized");
         return;
@@ -121,13 +121,13 @@ void ArduinoOTAClass::begin() {
         _hostname = tmp;
     }
     if(_mdnsEnabled){
-        MDNS.begin(_hostname.c_str());
+        MDNS.begin(_hostname.c_str(), tcpip_if);
         MDNS.enableArduino(_port, (_password.length() > 0));
     }
     _initialized = true;
     _state = OTA_IDLE;
 #ifdef OTA_DEBUG
-    OTA_DEBUG.printf("OTA server at: %s.local:%u\n", _hostname.c_str(), _port);
+    OTA_DEBUG.printf("OTA server at: %s.local:%u on iface %d\n", _hostname.c_str(), _port, tcpip_if);
 #endif
 }
 
diff --git a/libraries/ArduinoOTA/src/ArduinoOTA.h b/libraries/ArduinoOTA/src/ArduinoOTA.h
index ee8d589..7334f7c 100644
--- a/libraries/ArduinoOTA/src/ArduinoOTA.h
+++ b/libraries/ArduinoOTA/src/ArduinoOTA.h
@@ -64,7 +64,7 @@ class ArduinoOTAClass
     ArduinoOTAClass& onProgress(THandlerFunction_Progress fn);
 
     //Starts the ArduinoOTA service
-    void begin();
+    void begin(tcpip_adapter_if_t tcpip_if=TCPIP_ADAPTER_IF_STA);
 
     //Ends the ArduinoOTA service
     void end();

```

smuellener, 2018-07-09:
@dirkx what is the status of this issue? Is it currently possible to do OTA using ETH (e.g. LAN8720) rather than WIFI? Thank you

s00500, 2019-05-14:
Is there any reason this issue hasnt got any attention?

me-no-dev, 2019-05-14:
yeah. this is defunct... needs to be closed. MDNS has been running on all interfaces (including ETH) for a loong time. Closing now

closed 2019-05-14 by me-no-dev