espressif/arduino-esp32 #949: Deep Sleep not working for WiFi

Chop91, 2017-12-28:
### Hardware:
Board:							?ESP32 Devo Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?80Mhz?
Upload Speed:						?921600?

### Description:
I tried to modify the TimerWakeUp.ino from the examples, so that I should execute a HTTP GET request every 5 min. Between two requests the esp32 should go to deep sleep mode.
However, I managed to do the request, but no matter what I am doing, the WiFi seems to be turned on the whole time. I measure min 65mA, max 120mA and avg 85mA for example.
According to the datasheet (http://espressif.com/sites/default/files/documentation/esp-wroom-32_datasheet_en.pdf) I would expact something between 10µA and 150µA but not 65 mA..

Maybe someone can provide a simple sketch showing how to use deep sleep mode with WiFi?

I think I just reach "Modem-sleep", but how do I have to disable the WiFi after each update to reach "Deep-sleep"?

maxspad, 2018-06-02:
Having the same problem. Did you ever solve it?

lbernstone, 2018-06-02:
Did you try stopping the radios? https://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html#wifi-bt-and-sleep-modes

maxspad, 2018-06-02:
Yeah I stop the radios. Should i be releasing/stopping the Wifi library somehow?

lbernstone, 2018-06-02:
While deep_sleep clearly does not deinit WiFi (https://github.com/espressif/arduino-esp32/issues/1446), in my unscientific tests, I am able to get down to 10mA with the following code:
```
#include <WiFi.h>

#define NTP_SERVER "us.pool.ntp.org"
#define TZ_INFO "MST7MDT6,M3.2.0/02:00:00,M11.1.0/02:00:00" // Denver/N America
#define SLEEPTIME 1E7 //microseconds
  
void initFirst() {
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  delay(200);
  Serial.println(WiFi.localIP());

  configTzTime(TZ_INFO, NTP_SERVER);
  tm local;
  getLocalTime(&local, 10000); // wait up to 10sec to sync
}

void setup() {
  Serial.begin(115200);
  if (esp_sleep_get_wakeup_cause() != ESP_SLEEP_WAKEUP_TIMER) initFirst();
}

void loop() {
  time_t now=time(NULL);
  Serial.println(now);
  esp_sleep_enable_timer_wakeup(SLEEPTIME);
  esp_deep_sleep_start();
}
```

maxspad, 2018-06-03:
Unfortunately, the same code does not drop the current usage for me. I can tell from serial output that the the main processor DOES drop into deep-sleep mode and then resume, but there must be something on the board that once activated by WiFi doesn't turn off...

lbernstone, 2018-06-03:
What is your board?  May be an issue with something else on there using a lot more power than the esp32.  I know the SparkFun Thing32 had some issues https://forum.sparkfun.com/viewtopic.php?t=45931.

maxspad, 2018-06-03:
Solved it! The code is set up such that the board will only go into sleep mode after connecting to wifi. When I used the multimeter to measure its current, the internal resistance of the multimeter was too high, leaving too little left over for the board to connect to wifi - so it never entered sleep mode. I could measure current when I didn't turn on wifi, however, because the current draw was much smaller and there was no brownout before it entered sleep mode. By using the 10A ciruit on my multimeter (which is much less precise, unfortunately, but has much less internal resistance) I was able to see the board successfully enter deep sleep. 

@Chop91 maybe this was your problem?

closed 2019-08-15 by stale[bot]