espressif/arduino-esp32 #950: Cannot compile due to GLIBC_2.14 requirement

Shdwdrgn, 2017-12-29:
### Hardware:
Arduino: 1.8.5 (Linux), Board: "WEMOS LoLin32, 80MHz, 115200"

### Description:
I just did a fresh install of Arduino 1.8.5 on Debian Wheezy to build a new project.  However when I try to verify anything, including a new blank sketch, I get the following error:

/home/<user>/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.14' not found (required by /home/<user>/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++)
exit status 1

Note there is currently no actual ESP hooked up to serial, I'm simply trying to verify I have all the libraries in place to compile a project.  I was previously able to compile and flash code to an ESP-32S back in January of this year (under Arduino IDE 1.6.12).

Is there any specific reason why glibc 2.14 is required now when it wasn't previously?  Since this is an older system I don't have 2.14 available, I only have 2.13.

stickbreaker, 2017-12-29:
@Shdwdrgn sounds like the the last command of the install did not execute
```
python get.py
```
look over the installation instructions  [here](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md)

Chuck.

Shdwdrgn, 2017-12-29:
`$ git clone https://github.com/espressif/arduino-esp32.git esp32
Cloning into 'esp32'...
remote: Counting objects: 6872, done.
remote: Compressing objects: 100% (32/32), done.
Receiving objects:  52% (3574/6872), 36.94 MiB | 1.27 MiB/s   MiB/s   
git submodule update --init --recursiveReceiving objects:  57% (3918/6872), 48.0Receiving objects:  57% (3941/6872), 49.34 MiB | 1.27 MiB/s   
remote: Total 6872 (delta 10), reused 22 (delta 5), pack-reused 6832
Receiving objects: 100% (6872/6872), 103.63 MiB | 1.27 MiB/s, done.
Resolving deltas: 100% (3889/3889), done.
Checking out files: 100% (1272/1272), done.

$ cd esp32

$ git submodule update --init --recursive
Submodule 'libraries/BLE' (https://github.com/nkolban/ESP32_BLE_Arduino.git) registered for path 'libraries/BLE'
Cloning into 'libraries/BLE'...
remote: Counting objects: 216, done.
remote: Total 216 (delta 0), reused 0 (delta 0), pack-reused 216
Receiving objects: 100% (216/216), 125.71 KiB, done.
Resolving deltas: 100% (119/119), done.
Submodule path 'libraries/BLE': checked out 'f8fe9d7cdfb20caa54b70849826d1ac6e375ff78'

$ cd tools/

$ python get.py
System: Linux, Info: Linux-3.2.0-4-amd64-x86_64-with-debian-7.11
Platform: x86_64-pc-linux-gnu
Downloading xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Downloading mkspiffs-0.2.1-linux64.tar.gz
Done
Extracting mkspiffs-0.2.1-linux64.tar.gz
Renaming mkspiffs-0.2.1-linux64 to mkspiffs
Done`

That's the results from when I ran it during the initial install, but I don't see any error codes.  The page you linked to has the instructions I was following, however I skipped the first few parts because I had already previously install pip and pyserial.  Did I miss something?

palomarcoder, 2018-01-03:
I think you need the recursive option on the clone command

Shdwdrgn, 2018-01-03:
What would be missing without the recursive option?  The esp32 directory was populated with multiple levels of subdirectories, so recursion was obviously performed at some stage in the git command.

projectgus, 2018-01-04:
> Is there any specific reason why glibc 2.14 is required now when it wasn't previously? Since this is an older system I don't have 2.14 available, I only have 2.13.

The toolchain we released last month (1.22.0-75-gbaf03c2) was built from Debian Stretch and as a result we (accidentally) removed support for <glibc 2.14.

We're in the process of building and releasing a new toolchain build with restored glibc 2.13 support. In the meantime you could [download the toolchain version before this one](https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz) and manually copy it into the Arduino directory, and everything will *probably* work (the differences between -61 and -75 are C++ Exception support, which Arduino doesn't use, and an updated SPIRAM compatibility patch).

Shdwdrgn, 2018-01-04:
That's great that you found the problem, and I'm really glad to hear you haven't forgotten about those of us on older systems.  I'll keep an eye out for the new code as this isn't something I have to get working right this instant.

In the meantime I did find a work-around.  I actually built a VM with stretch and a barebones desktop install, from which I was able to compile the code and then transfer the .bin files back to my desktop for uploading to the ESP32.  A bit clumsy to work through, but my project is up and running.  It will get the job done until I can compile directly on my desktop again.  Thanks guys for all your work!

projectgus, 2018-01-04:
Glad you found a workaround.

FWIW, if your VM host supports some kind of USB or serial port passthrough then these should both work as well to flash directly from the VM (I do this all the time for testing various OS configurations).