espressif/arduino-esp32 #951: oneWire DS18b20 fix

stickbreaker, 2017-12-29:
Is anyone using Dallas OneWire devices.  I modified the standard Arduino OneWire library maintained by @PaulStoffregen.  So far It has returned accurate values from a couple of DS18b20's sharing the same pin(17).  If anyone is interested in test it with other devices/configurations feel free.  Constructive comments/ debug reports are welcome.

[stickbreaker/OneWire](https://github.com/stickbreaker/OneWire)

Chuck.

bbasil2012, 2017-12-29:
I just tried it.
Seems it run fine :) on my high load http server !!! 
WBR Basil.

stickbreaker, 2017-12-29:
@bbasil2012 Great!

CelliesProjects, 2017-12-29:
@stickbreaker Just swapped in your lib and enabled Dallas error logging in my project.
No more crc errors where before I had a crc error every 6-20 seconds.
Project uses 3 sensors max. also no problems there.

Great work!

palomarcoder, 2018-01-01:
I tried it, and it works for me.  Thanks!

stickbreaker, 2018-02-25:
No problems reported, I'm closing this.

closed 2018-02-25 by stickbreaker