espressif/arduino-esp32 #952: WiFi.SSID() does not provide the ssid string if not connected

szabonandi, 2017-12-29:
### Hardware:
Board:							D-duino-32
Core Installation/update date:			28/dec/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
WiFi.SSID() does not provide the ssid string if not connected.
ESP8266 behaves differently, it provides the ssid string even in connecting state. ESP32 does not.

SuperHam, 2018-01-01:
Could you link some specific code?

Are you able to use WiFi.RSSI to get the signal strength?

tablatronix, 2018-02-06:
wifista uses `esp_wifi_sta_get_ap_info` 

> Get information of AP which the ESP32 station is associated with

I use this for now to get the default SSID, when not connected.

```
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    return String(reinterpret_cast<const char*>(conf.sta.ssid));
```

mickeypop, 2018-02-12:
tablatronix ; your pre connect solution works but you forgot to post the include to make it work

`		#include <esp_wifi.h>
`

all said, a great help

tablatronix, 2018-02-12:
My bad I use it in a library so forgot

mickeypop, 2018-02-13:
I use my own, like yours but a variant let you call what you need easily add
anything in the conf.sta. struct say bssid etc..   Best of all it works in any state of the WiFi in IDF or Arduino .

```
// Requires; #include <esp_wifi.h>
// Returns String NONE, ssid or pass
// ie   String var = getSsidPass( "pass" );  //returns pass phrase
String getSsidPass( String s )
{
  String val = "NONE";  // if invalid, returns "NONE"
  s.toUpperCase();
  if( s.compareTo("SSID") == 0 )
  {
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    val = String(reinterpret_cast<const char*>(conf.sta.ssid));
  }
  if( s.compareTo("PASS") == 0 )
  {
    wifi_config_t conf;
    esp_wifi_get_config(WIFI_IF_STA, &conf);
    val = String(reinterpret_cast<const char*>(conf.sta.password));
  }
  return val;
}
```

beegee-tokyo, 2018-02-14:
Nice finding. I added this info together with a code example into the [Wiki](http://desire.giesecke.tk/index.php/2018/02/14/get-stored-wifi-ap-and-sta-info/).

SuperHam, 2018-02-14:
@mickeypop From your code above, when you call:
```
    esp_wifi_get_config(WIFI_IF_STA, &conf);
```
does this turn the radio on/broadcast or is it grabbing the past stored values without powering the unit?

I am working on a project and a major concern is power consumption and I'm trying to avoid broadcasting when not necessary.

mickeypop, 2018-02-14:
It simply uses ta defined union/struct to get the stored info when the radio is OFF or ON

You just have to remember to add the **#include <esp_wifi.h>** so the union/struct is loaded or it will not know where to get it from.

szabonandi, 2018-02-17:
I repeat the experiment, it seems that the problem can be workarounded if some delay is added before acquiring the SSID string (and it applies to RSSI as well).

This is my test code:

```
#include <WiFi.h>

const char* ssid     = "Telekom-285565";
const char* password = "secret";

void setup()
{
  Serial.begin(115200);
  delay(10);

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  int msec = 0;

  while (WiFi.status() != WL_CONNECTED) {
    delay(100);
    Serial.println(String(msec) + " msec: SSID=" + WiFi.SSID() + " status=" + WiFi.status() + " RSSI=" + WiFi.RSSI());
    msec = msec + 100;
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop()
{
  delay(5000);
}
```

this is the serial output, after 700msec the values appears:

```
Connecting to Telekom-285565
0 msec: SSID= status=6 RSSI=0
100 msec: SSID= status=6 RSSI=0
200 msec: SSID= status=6 RSSI=0
300 msec: SSID= status=6 RSSI=0
400 msec: SSID= status=6 RSSI=0
500 msec: SSID= status=6 RSSI=0
600 msec: SSID= status=6 RSSI=0
700 msec: SSID=Telekom-285565 status=6 RSSI=-88
800 msec: SSID=Telekom-285565 status=6 RSSI=-81
900 msec: SSID=Telekom-285565 status=6 RSSI=-81
1000 msec: SSID=Telekom-285565 status=6 RSSI=-81
1100 msec: SSID=Telekom-285565 status=6 RSSI=-79
1200 msec: SSID=Telekom-285565 status=6 RSSI=-81
1300 msec: SSID=Telekom-285565 status=6 RSSI=-80
1400 msec: SSID=Telekom-285565 status=6 RSSI=-82
1500 msec: SSID=Telekom-285565 status=6 RSSI=-83
1600 msec: SSID=Telekom-285565 status=6 RSSI=-80
1700 msec: SSID=Telekom-285565 status=6 RSSI=-80
1800 msec: SSID=Telekom-285565 status=6 RSSI=-79
1900 msec: SSID=Telekom-285565 status=6 RSSI=-80
2000 msec: SSID=Telekom-285565 status=6 RSSI=-80
2100 msec: SSID=Telekom-285565 status=6 RSSI=-80
2200 msec: SSID=Telekom-285565 status=6 RSSI=-78
2300 msec: SSID=Telekom-285565 status=6 RSSI=-79
2400 msec: SSID=Telekom-285565 status=6 RSSI=-80
2500 msec: SSID=Telekom-285565 status=6 RSSI=-79
2600 msec: SSID=Telekom-285565 status=6 RSSI=-78
2700 msec: SSID=Telekom-285565 status=6 RSSI=-79
2800 msec: SSID=Telekom-285565 status=3 RSSI=-79

WiFi connected

WiFi connected
IP address: 
192.168.1.62
```

tablatronix, 2018-02-17:
Or you can use events and wait for it, but that is only if you want to actually connect to wifi.
Also note that this will probably change when persistent is actually implemented, as of now its just a flag that does nothing if i read the code right.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

dachshund-digital, 2019-03-24:
Events idea does not seem to work to get SSID during a connection event. For example the following code returns empty string for SSID during a connect event when WiFiMulti.run is searching for best network.

    case SYSTEM_EVENT_STA_START:
      
      //
      sprintf(theMessage, "ESP32 Station Start %s", WiFi.SSID().c_str());
      //strcpy(theMessage, "ESP32 Station Start");

Returns empty string for WiFi.SSID().

atanisoft, 2019-03-24:
@dachshund-digital at that point the SSID is not available as it hasn't been connected to yet. You will want the connected event instead.

dachshund-digital, 2019-03-25:
Ah, that leads to an additional question... is there an event or point in time that I can tell which AP a connection is being attempted on or to?  I have a WiFiMulti.run loop that has 2 or 4 APs as connection candidates, and was hoping I could determine which AP is being considered while the .run sequence is active.  There does not seem to be an obvious way to determine this?  I guess I could fore go using WiFiMulti... write my own logic then I might be able to track which AP is hit, using the signal strength as criteria, but really that is what .run is doing already.  Maybe a feature request is to have new event, that might feed that information to the event handler?  Connection Started event fires for each pass in .run, so I guess it could be done?

atanisoft, 2019-03-25:
@dachshund-digital that would be up to the WifiMulti library to expose. There is no event raised for connecting, only one for after a connection has been made.

I'd say modify wifi multi and submit a PR for it.