espressif/arduino-esp32 #953: GPIO input mode with softAP

Aswinvt, 2017-12-29:
### Hardware:
Board: ESP32 Dev Module
IDE name: Arduino IDE
Flash Frequency: 40Mhz
Upload Speed: 115200

### Description:
I am trying to read analog value from some sensors using ESP32 and that values are needed to be viewed in a browser, for that I have made ESP32 as AP, but after setting it as AP the only raw analog read value that I get is 4095.

### Test code:
```cpp

#include <WiFi.h>

 
const char *ssid = "TestAP";
const char *password = "testappwd";
 
void setup(){
  Serial.begin(115200);
  pinMode(0, INPUT);
  WiFi.softAP(ssid, password);
  Serial.println();
  Serial.print("IP address: ");
  Serial.println(WiFi.softAPIP());
}
 
void loop(){
  float IN = analogRead(0);
  Serial.println(IN);
  delay(1000);
}
```

Above is a test code to verify this issue after setting the ESP32 as AP the analogRead function only returns a value of 4095 only, otherwise it gives correct values.

I don't know whether I am missing out anything. Please help me out with this problem. 

Thanks.

beegee-tokyo, 2017-12-29:
You might get this problem because [ADC2 channel cannot be used when WiFi is in use](https://github.com/espressif/arduino-esp32/issues/440) or [Conflict Between Wifi & analogRead()? ](https://github.com/espressif/arduino-esp32/issues/102). Try to use another ADC channel. I am using the following (just an extract of my code) to read the analog value from a LDR. No AP here though, I am using WiFi in STA mode:
```cpp
/** Pin number for LDR analog input pin */
int ldrPin = 36;
/** LDR light value = 0 if not updated */
long newLDRValue = 0;

void setup(void) {
  // Initialize analog port for LDR
  pinMode(ldrPin,INPUT);
  adcAttachPin(ldrPin);
  analogReadResolution(11);
  analogSetAttenuation(ADC_6db);
}

void loop(void) {
  // Read analog value of LDR
  newLDRValue = analogRead(ldrPin);
}
```

Aswinvt, 2017-12-29:
Thank you for the reply. After changing to the ADC1 analog read is working without any problem.

closed 2017-12-29 by Aswinvt