espressif/arduino-esp32 #956: esp32 bluetooth mac scan

mahdikan, 2017-12-30:
hello everyone
i want to scan for bluetooth devices near the esp32 to find thats mac, but i couldn't find example in arduino
please help me if you have an example.
thanks
mahdi

mahdikan, 2018-01-03:
any body?

copercini, 2018-01-03:
For now there is just a IDF example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bt_discovery but you can run it on arduino with small changes

mahdikan, 2018-01-03:
hi copercini 
thanks alot for your mention, but i can't change it to arduino , i copied the code in arduino but there are som errors,
may i asked you to explain changing the code?

copercini, 2018-01-03:
Humm, the BT core of Arduino is a bit outdated in relation to IDF, so for now you can use [arduino as IDF component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md) or wait a bit to @me-no-dev update the arduino core with the last IDF version

mahdikan, 2018-01-03:
how long should i wait to update arduino core?

beegee-tokyo, 2018-01-04:
@copercini, the [example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bt_discovery) will not work, as it requires esp-idf V3.0 at least. At least on PlatformIO we are on esp-idf V2.50200.0

beegee-tokyo, 2018-01-04:
@mahdikan are you searching for Bluetooth devices or for BLE devices?
The [BLE scan](https://github.com/nkolban/ESP32_BLE_Arduino/tree/f8fe9d7cdfb20caa54b70849826d1ac6e375ff78/examples/BLE_scan) example is working.

I have it running on my ESP32, but it finds only BLE devices (of course).

mahdikan, 2018-01-06:
thanks to beegee yes the ble scan is working but i want to scan smartphones and cars bluetooth mac

c0ryd, 2018-01-20:
What are you trying to do with the MAC address?  If you are attempting to identify a specific device (like a presence detector), that’s probably not going to work for you. Many devices have a rolling MAC address, so your sketch might work for ~15 minutes, then break. 
What you might want to go after is the Identity Resolving Key, but as to how to do that I haven’t been able to sort that out just yet.

mahdikan, 2018-01-20:
hello c0ryd 
yes i want to identify  some devices MAC for calculating the repeated costumers in a shop during one month , do you know the rolling MAC address algorithm ? or do you know the way for identify the bluetooth mac?
thanks

c0ryd, 2018-01-20:
I’m far from an expert, but as far as I know the algorithm for the Mac addresses won’t be public, as the whole purpose is for privacy, so your device can’t be tracked like that. 
From what I understand, there is an “identity resolving key” but that’s only available if the device has been paired and bonded with your “sniffer”. 
Perhaps track people instead via WiFi? You would be able to track reconnects easier, and who doesn’t want free WiFi access?  You might even use an iBeacon to push a message to your customers with your WiFi password when they enter the store, to get more people to connect.

closed 2018-04-19 by everslick