espressif/arduino-esp32 #957: ESP32 with boards manager in Arduino IDE 1.8.5

patrickelectric, 2017-12-31:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			30/dez/2017
IDE name:					       Arduino IDE 1.8.5

### Description:
Is it possible to install esp32 package in Arduino IDE ?
Steps:
```bash
arduino --pref boardsmanager.additional.urls="https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json" --save-prefs
arduino --install-boards esp32:esp32
```

Output:

```
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Downloading platforms index... 
Downloading platforms index... Downloaded 0kb of 197kb.
Downloading platforms index... Downloaded 5kb of 197kb.
Downloading platforms index... Downloaded 11kb of 197kb.
Downloading platforms index... Downloaded 21kb of 197kb.
Downloading platforms index... Downloaded 27kb of 197kb.
Downloading platforms index... Downloaded 33kb of 197kb.
Downloading platforms index... Downloaded 41kb of 197kb.
Downloading platforms index... Downloaded 44kb of 197kb.
Downloading platforms index... Downloaded 47kb of 197kb.
Downloading platforms index... Downloaded 50kb of 197kb.
Downloading platforms index... Downloaded 55kb of 197kb.
Downloading platforms index... Downloaded 59kb of 197kb.
Downloading platforms index... Downloaded 65kb of 197kb.
Downloading platforms index... Downloaded 68kb of 197kb.
Downloading platforms index... Downloaded 73kb of 197kb.
Downloading platforms index... Downloaded 78kb of 197kb.
Downloading platforms index... Downloaded 86kb of 197kb.
Downloading platforms index... Downloaded 92kb of 197kb.
Downloading platforms index... Downloaded 95kb of 197kb.
Downloading platforms index... Downloaded 102kb of 197kb.
Downloading platforms index... Downloaded 107kb of 197kb.
Downloading platforms index... Downloaded 112kb of 197kb.
Downloading platforms index... Downloaded 123kb of 197kb.
Downloading platforms index... Downloaded 133kb of 197kb.
Downloading platforms index... Downloaded 139kb of 197kb.
Downloading platforms index... Downloaded 150kb of 197kb.
Downloading platforms index... Downloaded 153kb of 197kb.
Downloading platforms index... Downloaded 157kb of 197kb.
Downloading platforms index... Downloaded 165kb of 197kb.
Downloading platforms index... Downloaded 169kb of 197kb.
Downloading platforms index... Downloaded 173kb of 197kb.
Downloading platforms index... Downloaded 176kb of 197kb.
Downloading platforms index... Downloaded 181kb of 197kb.
Downloading platforms index... Downloaded 183kb of 197kb.
Downloading platforms index... Downloaded 188kb of 197kb.
Downloading platforms index... Downloaded 191kb of 197kb.
Downloading platforms index... Downloaded 197kb of 197kb.
Downloading platforms index... 
Downloading platforms index... Downloaded 0kb of 0kb.
Downloading platforms index... Downloaded 0kb of 0kb.
Downloading platforms index... 
Downloading platforms index... Downloaded 0kb of 6kb.
Downloading platforms index... Downloaded 1kb of 6kb.
Downloading platforms index... Downloaded 4kb of 6kb.
Downloading platforms index... Downloaded 6kb of 6kb.
Downloading platforms index... 
java.lang.NullPointerException
	at java.util.Collections.sort(Collections.java:175)
	at processing.app.Base.<init>(Base.java:310)
	at processing.app.Base.main(Base.java:144)
```

It's not possible to find esp32 boards using IDE with Additional Boards Manager URLs.

lbernstone, 2018-01-06:
https://github.com/espressif/arduino-esp32#installation-instructions

patrickelectric, 2018-01-07:
@lbernstone this is an absolute method, the new Arduino way to do is using the `boardsmanager.additional.urls`,

jjimmy-S, 2018-03-27:
I added the specified url but nothing happens...

eren23, 2018-03-27:
I don't really get how to make it work, can you explain it a little bit detailed.

> arduino --pref boardsmanager.additional.urls="https://raw.githubusercontent.com/espressif/arduino-esp32/master/package/package_esp32_index.template.json" --save-prefs
arduino --install-boards esp32:esp32

Testato, 2018-03-27:
Duplicate
https://github.com/espressif/arduino-esp32/issues/515

copercini, 2018-04-21:
Closed as duplicated of https://github.com/espressif/arduino-esp32/issues/515

Let's concentrate the discussion on https://github.com/espressif/arduino-esp32/issues/515 to keep the thing organized =)

closed 2018-04-21 by copercini