espressif/arduino-esp32 #961: xtensa-esp32-elf-g++": file does not exist Error compiling for board Adafruit ESP32 Feather.

perigalacticon, 2018-01-01:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?Adafruit ESP32 Feather?
Core Installation/update date:			?12/31/17?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I get this error when compiling any sketch for the ESP32 Feather:
`exec: "C:\\Users\\Perigalacticon\\Documents\\Arduino\\sketches\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist
Error compiling for board Adafruit ESP32 Feather.`

I tried reinstalling the core 4 times, the directions say that the sketchbook directory is:

> Sketchbook Directory: Usually C:/Users/[YOUR_USER_NAME]/Documents/Arduino 

But this is not true, this is not the usual sketchbook directory, usually it is: 
 C:/Users/[YOUR_USER_NAME]/Documents/Arduino/sketches.  

Anyway I can't compile any sketch if installed to either directory, it gives the error immediately that it can't find the files and I have restarted and reinstalled everything several times.  Can you please help me.  I have a standard up-to-date IDE installation on Windows10 64 bit.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

WereCatf, 2018-01-01:
Did you run get.exe?

perigalacticon, 2018-01-01:
Yes.  Also now I cannot upload to any esp8266 either by FTDI, only OTA.  There are missing files here also, this is the error:
`java.io.IOException: Cannot run program "C:\Users\Perigalacticon\Documents\Arduino\sketches\hardware\esp8266com\esp8266/tools/esptool/esptool.exe": CreateProcess error=2, The system cannot find the file specified`

How do I reinstall the esp8266?  I used the boards manager.  And Esp32 will still not compile.

perigalacticon, 2018-01-01:
FYI I fixed the ESP8266 section by deleting the ESP8266com directory and restarting now it compiles again and uploads for the ESP8266, but I have the same error for the ESP#2, it doesn't find the files.

perigalacticon, 2018-01-04:
So I tried installing this to both the \Documents\Arduino\sketches\hardware\espressif\esp32 directory and the \Documents\Arduino\hardware\espressif\esp32 directory.  When installed in the non-sketches directory I don't get a list of ESP32 boards in the menu.  I have installed in both locations twice and the same thing happens.  I deleted the espressif directory in each before proceeding.  For the sketches directory install I get the boards list but nothing will compile.  I have a 64bit machine so I also have an Arduino directory at C:\Program Files (x86)\Arduino\.  Is it supposed to be installed here instead?  I really appreciate your help I am looking forward to using the board.

RbotJ, 2018-01-21:
Was running into the same issue with xtensa-esp32-elf not appearing in my tools directory. It appeared (almost magically) after I uninstalled the ide (1.8.2) and installed the most recent version (1.8.5). I have now successfully uploaded and tested the "touch" sample sketch.

mpo881, 2018-01-30:
I have been going around and around with this for days now. I stated with Arduino 1.6.5 and could not get a sketch to compile to Wemos D1 Mini. So, I upgraded to Arduino 1.8.5. Now I am getting the same error as the OP reports. I have tried all the same remidies without success. I have no clue what the fix is!

mpo881, 2018-01-30:
I found this and ran all the python script. That fixed the wemos, no luck with the ESP32 Dev board though
https://github.com/espressif/arduino-esp32/issues/581

ron15, 2018-02-09:
I have encountered the same problem and solve it by running the get.exe again but as administrator.

Here's what i've done:
Right Click "get.exe" and select "run as administrator".

Hope this solves your issue.

mpo881, 2018-02-12:
@ron15 Ill give it a try. Thank you for your reply

mimansamaheshwari, 2018-02-16:
I read the discussion regarding and also followed to get rid of this problem but still can't find reason why this is happening. I am also getting this compiling error,
"C:\\Program Files (x86)\\Arduino\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++": file does not exist".
I ran get.exe as administrator as well but that too didn't work.
Can anyone help? please!

ron15, 2018-03-06:
hmmm... its really odd.. are you connected to the internet while doing this? I think get.exe downloads the files needed. In my case, i just assumed that windows doesnt allow the program to write on the directory that's why I tried to run it as administrator.

The get.exe must be in the "C:\Program Files (x86)\Arduino\hardware\espressif\esp32" i suppose. 

Also check the directory "C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" if it exist.

alexvaughn23, 2018-03-13:
Running get.exe as an Administrator worked for me to solve the xtensa problem in compiling. Also I extracted the core directly into the C:\Program Files(86x)\....expressif, then renamed the folder to esp32.
Just ran my first blink upload. Officially good to go. Wish it was this easy on Mac... Let me know if you find a solution for the ESP32 for High Sierra on a Macbook.

gautam-chebrolu, 2018-03-22:
I was having the same issue as OP for a while and the solution that worked for me was to make sure all the slashes are in the same direction when cloning the repository from Git. I originally had the source directory as:
`"C:\Users\[name]\Documents\Arduino\sketches\hardware/espressif/esp32"`
and it would look like it would work because most of the files would load and the libraries would be available in Arduino, but "get.exe" would take forever and just not finish executing and there were issues compiling in the Arduino IDE. Then I changed it the source directory to: 
`"C:/Users/[name]/Documents/Arduino/sketches/hardware/espressif/esp32"` and then everything started working.

thkinker, 2018-04-10:
To make the get.exe to create the xtensa-esp32-elf folder on my Windows machine I had to change the 'Enable CyberCapture' to 'allow me to decide' in my AVG settings. Run the get.exe as administrator and now it's working as expected.

KevinLod, 2018-05-10:
@ron15  , would you please tell me how to run get.exe

cyhuvraak, 2018-05-30:
@KevinLod check espressif\esp32\tools folder

Vadim981, 2018-12-23:
Когда запускаєш get.exe нужно не закрывать окно програми get.exe, а подождать полной загрузки. Все файли автоматически установляться у папки

AMC-Electronics, 2019-02-23:
> @KevinLod check espressif\esp32\tools folder

Thx this solved the problem for me.

UtkarshChandel, 2019-09-25:
- [ ] 1.Just Clone the repo **[https://github.com/espressif/arduino-esp32](url)**

- [ ] 2. Copy the contents to "C://Users/<UserName>/documents/arduinio/hardware/espressif/esp32/"

- [ ] 3. Goto **[https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-59.zip](url)**  and download the zip file

- [ ] 4. Extract the zip file in the tools folder of esp32 i.e.  "C://Users/<UserName>/documents/arduinio/hardware/espressif/esp32/tools/"

- [ ] 5. Run get.exe as administrator.

_Hope this helps._

babursenturk, 2020-05-14:
This is totally SSL problem. Using another internet connection without any firewall restriction will resolve the problem. Probably you have strong firewall rules. (Maybe your company uses Fortinet.) I'm sure you can't send https request through GIT not only for this but also many others. In order to test it; open a GIT GUI and clone a repostry from Hub with have https prefix. If fetching is not start and give an error then probably you have firewall.

anuparp-b, 2020-08-18:
In my case,
I found errors
fork/exec C:\Users\xxxx\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++.exe: The file or directory is corrupted and unreadable.
Error compiling for board Heltec WiFi LoRa 32(V2).

At  C:\Users\xxxx\AppData\Local\Arduino15\packages, there are two folders.

- esp32
- Heltec-esp32

Which their sub-folders have "xtensa-esp32-elf-g++.exe" . I guessed that Arduino IDE is confused between two paths when compile programs.

Therefore, I tried to move the folder of esp32 to my desktop folder. The problem is solved that i can run Heltec-esp32 agian.

thaynapatricio, 2021-11-22:
Hey Guys! 
This video soved my problem in VSCode
Error: exec:" / bin / xtensa-esp32-elf-g ++

www.youtube.com/watch?v=d1D83G-PpqM

Murkas94, 2022-11-09:
I had this issue after updating the esp32-library through VS-Code with the Arduino-Plugin (from Microsoft, not PlatformIO).
I fixed it by using the Arduino-IDE to uninstall and then install the esp32-library again.