espressif/arduino-esp32 #962: MDNS.addService not working

natcl, 2018-01-02:
Please fill the info fields, it helps to get you faster support ;)
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			
IDE name:							Arduino 1.8.5 on MacOS
Flash Frequency:					?80Mhz?
Upload Speed:						

### Description:
Using the mDNS_Web_Server example sketch, I can't get the addService method to work.  The sketch runs fine, I can access my ESP32 using the esp32.local URL but it doesn't get advertised on the network when checking with different mdns browsers.  Similar code works with the esp8266.  Also browsing services seems to work.

baggior, 2018-02-03:
I confirm this problem

baggior, 2018-02-03:
#387

beegee-tokyo, 2018-02-04:
Works without problems for me, but I am not using MDNS.begin, instead I am using ArduinoOTA to start the mDNS service.
```cpp
ArduinoOTA.setHostname("ESP32-Test-2481CE9C");
ArduinoOTA.begin();
// addd some more service text
MDNS.addServiceTxt("_arduino", "_tcp", "service", "MHC");
MDNS.addServiceTxt("_arduino", "_tcp", "type", "TestBoard");
MDNS.addServiceTxt("_arduino", "_tcp", "id", "ESP32-Test");
```
Result of Service Browser on my Android tablet:
[image]

natcl, 2018-02-04:
But that’s to advertize the Arduino service. If you want to advertize a generic service (like OSC over UDP) it doesn’t work.

baggior, 2018-02-05:
No the service is not shown even in Service Browser forAndroid
The same code works for ESP8266 but not for ESP32
I did 

WiFi.setHostname(..)
MDNS.begin()
MDNS.addService(..)

beegee-tokyo, 2018-02-05:
Maybe it works for me, because I am using the "staging" version, but when I do 
```cpp
	WiFi.setHostname("ESP32-Test-mDNS");
	MDNS.begin("ESP32-Test-mDNS");
	MDNS.addService("_osc","_udp",4500);
	MDNS.addServiceTxt("_osc","_udp","does","it");
	MDNS.addServiceTxt("_osc","_udp","work","?");
	MDNS.addServiceTxt("_osc","_udp","yes","!");
```
The result in Android service browser is
[image]

I see that you are saying you use `MDNS.begin();`. That throws a compilation error for me. MDNS.begin() requires a parameter which is the hostName.
```log
src\ota.cpp: In function 'void activateOTA(const char*, const char*)':
src\ota.cpp:126:13: error: no matching function for call to 'MDNSResponder::begin()'
MDNS.begin();
^
In file included from src\setup.h:20:0,
from src\ota.cpp:1:
C:\users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\ESPmDNS\src/ESPmDNS.h:57:8: note: candidate: bool MDNSResponder::begin(const char*)
bool begin(const char* hostName);
^
C:\users\beegee\.platformio\packages\framework-arduinoespressif32\libraries\ESPmDNS\src/ESPmDNS.h:57:8: note:   candidate expects 1 argument, 0 provided
*** [.pioenvs\esp32dev\src\ota.o] Error 1
```
And again, I am on the latest staging version:
[image]

beegee-tokyo, 2018-02-05:
Ok, I started comparing the ESPmDNS.cpp (ESP32) with the ESP8266mDNS.cpp (ESP8266) and found some interesting differences.
On the ESP32 version you have to make sure
- that the service name is lower case
- that the service name and protocol starts with an '_' e.g. '_udp'    

On the ESP8266 version the library takes care of
- copying service name to lower case
- adding the '_' in front if necessary.

So maybe try 
- to change your service name into all lower case and add the '_' in front
- to change your protocol name into all lower case and add the '_' in front

It nether bothered me because I did the same all my ESP8266 projects and ESP32 projects already, so I never saw the problem.

@me-no-dev - Maybe ESPmDNS needs some update to be 100% compatible with the ESP8266 version.

xinort, 2018-02-05:
Doesn't work for me either. And hasn't for quite a while.

beegee-tokyo, 2018-02-05:
I tried it on 
- Arduino IDE with the lastest master version of the platform
- PlatformIO/Visual Studio Code with the lastest master version of the platform
- PlatformIO/Visual Studio Code with the lastest staging version of the platform

I tried it with the ESP32 example codes
- mDNS_Web_Server.ino
- mDNS-SD_Extended.ino

In the example codes I only made the changes from **`"http", "tcp"`** to **`"_http", "_tcp"`** to get it working.

One thing I found out is that after flashing, the mDNS advertise sometimes couldn't be found immediately. Then I had to do a hard reset (push reset button or power off/on cycle). After a hard reset it worked always.

Can you give me a code excerpt that shows the initialization and adding of services?

baggior, 2018-02-05:
@beegee-tokyo I've added the "_" prefix to service and protocol and now it works..
the service is published and visible in Android service browser..

but the query method doesn't work since MDNS.query cannot find any result..( with or without the prefix)

Anyway there is a bug or a serious  inconsistency, since in the query method code the  "_" prefix is added anyway while int the addService method the prefix is required to be provided in the input parameter..

beegee-tokyo, 2018-02-05:
> Anyway there is a bug or a serious inconsistency, since in the query method code the "_" prefix is added anyway while int the addService method the prefix is required to be provided in the input parameter..

Saw this as well. Inconsistent and in general the ESP8266 and ESP32 mDNS should work the same.

But happy I could at least with the advertising.
I tried the query example mDNS-SD_Extended.ino and it worked for me. It found all mDNS services in my system. Will copy my working code into a Gist and post the link tomorrow.

baggior, 2018-02-05:
@beegee-tokyo so can you fix that all?
commit and open a pull request?

beegee-tokyo, 2018-02-05:
Will try in the next days.

beegee-tokyo, 2018-02-06:
@baggior pull request sent, now it is up to @me-no-dev to merge it.

baggior, 2018-02-06:
@beegee-tokyo thank you anyway the query() method doesnt works for me with or without '_'
I'm using stagine too

can you please check also this bug and try to fix it in the same PR #1085?

beegee-tokyo, 2018-02-06:
@baggior I successfully tested the query with "standard" services like http, workstation or SSH. Didn't try it with UDP port and "special" service.
Will setup a device as `_osc._udp` and see what I can do.
Btw. query() does **not** find a mDNS advertisement registered on the same device, only other devices.

baggior, 2018-02-06:
Really 
my test was to find the same device so  I have to make another test ..

please add a note in the source as a **comment** about that then...

beegee-tokyo, 2018-02-06:
I think that is intended behavior. I setup an **_`_osc._udp`_** mDNS service on an **ESP8266** and an **ESP32**.   
Then I started a query on both devices.
Each device shows only the **_`_osc._udp`_** of the other device **not it's own**.
So behaviour of ESP32 and ESP8266 mDNS query are the same.

beegee-tokyo, 2018-02-17:
@baggior I do not make any update to the mDNS services after I fired them up and they still show after hours. But I do not use the _workstation._tcp at the moment. As this is the one you see after the others disappeared, it might be connected to this service. How do you initiate that service? Are you using the `enableWorkstation(wifi_interface_t interface)` function from the ESPmDNS library or something else?

sorenhl, 2018-06-01:
@baggior Did you ever find a solution for this?

Tried the example code to align it with the thread:
`WiFi.setHostname("ESP32-Test-mDNS");
MDNS.begin("ESP32-Test-mDNS");
MDNS.addService("_osc", "_udp", 4500);
MDNS.addServiceTxt("_osc", "_udp", "does", "it");
MDNS.addServiceTxt("_osc", "_udp", "work", "?");
MDNS.addServiceTxt("_osc", "_udp", "yes", "!");`

With ESP8266 and this works perfectly, but not with ESP32 (ESP-Wroom-32).
If i use Zeroconf (C#) to discover services my esp8266 shows up with the correct name but ESP32 shows up with name "_osc" and dissapears after like 10 seconds. If I hard-reset it, the same thing happens, it shows up as "_osc" in display name and dissapears again.

Any ideas?

versamodule, 2018-09-22:
FWIW, i was also having the same trouble. I was using Firefox this whole time. I tried it with Opera and using http://esp32.local/ it worked perfect.

LuckyLuzz, 2018-11-15:
```
  MDNS.addService("_http", "_tcp", 80);
  MDNS.addServiceTxt("_http", "_tcp", "board", "ESP32");
```
Works on ESP32 devkit V4 with Opera but not with Firefox or Edge.

evert-arias, 2018-11-29:
Adding the _  in front, worked for me
` MDNS.addService("_http", "_tcp", 80);`

closed 2019-08-15 by stale[bot]

ggilestro, 2019-08-20:
I finally found a workaround for this.
The issue must be linked with the powersave functions because disabling power save on the WIFI solves it for me.

```
    // We remove power save functions on WiFi to try and solve the mDNS disappearance issue
    WiFi.setSleep(false);
```

(Here is the full [code](https://github.com/gilestrolab/ethoscope/blob/57b01fa6d25e656c21cb8d8e620e9fd801247080/scripts/esp32_environmental_sensors/esp32_environmental_sensors.ino))

Without that line, the mDNS disappears after 6-12 hours.
With that line, it has been running for 48 hours so far.